opentx/opentx #5325: intermittent lost of control  on QX7 with 2.2 release candidate from about 6 months old.

mstrens, 2017-11-07:
A friend is using a QX7 and his running openTX 2.2 (a release candidate from 6 monts ago).
His SD card contains the files of 2.2.0  (released version) and so it is not compatible with his openTx version.
So each time he powers up the radio, he get a warning message about incorrect SD card version but after pressing the rotary wheel, he can use the radio.

Some days ago, he still had a crash with his plane because he lost control of servos while flying.
When this occured, the LED on the TX power switch becomed red and the display gave a message about a wrong SD card version.

After the plane crash, the Tx worked again.

He reported me that in the last days he got strange behaviours (red led again and once a blank display). Still I was not present to note this.

Yesterday, when he explains me those behaviour, I took the Tx while TX and Rx where powered on and I begin to gave some little shocks on the TX.
Suddently, the led of powerswitch became red and the display gave the message about wrong SD card version. At this time, I moved the sticks but the servos did not react. 
When I pressed on the rotary wheel, the servo moved again.
So this seemed to be the same issue as some days before.

I tried to reproduce the issue in 2 ways : giving some shocks on the Tx and (after opening the tx) moving up and down some cables (battery , between main board and power on board,...). Sorry but I was not able to reproduce the issue.

So I suspect that in some cases (?), a watchdog occurs and then openTx blocks if the sd card version is not the rigth one (until the message is acknowledge).

If it is so, it would be better that a wrong SD card version does not stop the Tx in case of a watchdog event.

Do you have an idea how a shock on Tx could cause a watchdog event? Could it be that it is linked to the battery used for RTC?

kilrah, 2017-11-07:
There is likely a bad contact somewhere on a power-related connection, or a broken battery wire.
OpenTX detects watchdog resets and skips the warnings, but a power disconnect may not get caught. 

Also he should be using release firmware, release candidates were not supposed to be used beyond testing.

closed 2017-11-07 by kilrah

mstrens, 2017-11-09:
I do not understand how a power disconnect could cause this issue because when I disconnect/reconnect  the battery when Tx was powered on, the Tx does not reboot automatically.

On the other side, if a watchdog reset skips all warnings, how can it be that the Tx freezed on the SD card version warning?

kilrah, 2017-11-09:
Because as I mentioned it likely wasn't a watchdog reset.

I doubt you can reproduce a typical intermittent connection lasting a few milliseconds by manually disconnecting the battery.

Update firmware, reseat the connections, and IF you still have the issue then report back.