opentx/opentx #5326: LCD type and stick Gain values lost after transfering data to ersky9x

capitaineflam25, 2017-11-07:
Hi,

i have an T9X+sky9x board+Optrex LCD+Aurora gimbals (OpenTX 2.2.0)

It's OK when i manually set up the screen type and the gain values from the radio, but every time i transfer data from Companion to my TX : those settings are lost.

It may be useless to access those settings from Companion, but keeping them after a transfer would be nice.

[image]

schwabe, 2017-11-07:
Are you using OpenTX or ersky9x? Your bug title and description are mismatching. 

Note that ersky9x on the radio and OpenTX companion is not going to work.

capitaineflam25, 2017-11-07:
Sorry for the wrong title, i've changed it.

I confirm i'm using OpenTx firmware : 
[image]

commit e2858dbb (2017-11-08): [companion] Some fields missing for Sky9x (fix #5326)
closed 2017-11-08 by kilrah