opentx/opentx #5329: SD Card brower issue

LapinFou, 2017-11-07:
Hi folks,

When cleaning-up my **FIRMWARE** folder on my SD card, I discovered than I couldn't open some text files. I didn't tried yet on my radio, so I report the bug for Companion only.
Not sure if it is a known limit or a a bug.

Step to reproduce the issue:
1.  download this archive → [FIRMWARE.zip](https://github.com/opentx/opentx/files/1450628/FIRMWARE.zip)
2. Unzip into the root of the SD card.
3. Use the SD card browser:
**_X4R_cppm.txt** can be open.
**_X4R-X4RSB_cppm.txt** cannot be open.

kilrah, 2017-11-08:
Path too long.

closed 2017-11-08 by kilrah

LapinFou, 2017-11-08:
OK. So, this is not a "bug" ?

powercycle, 2017-11-08:
I think it is a limitation of the file system.

kilrah, 2017-11-08:
Max path length for text files is set to 40 chars to avoid wasting useful memory.

LapinFou, 2017-11-09:
OK. Got it. A popup warning message would be nice. But, it is not really important.