opentx/opentx #5330: X7/X12: Support for additional switch inputs

kilrah, 2017-11-08:


1Sven, 2017-11-08:
I hope the topic will not be forgotten. I'm very interested in that...

kilrah, 2017-11-08:
No, but won't happen before next major version.

Crunch69, 2017-11-22:
will this allow the addition of 2 extra switches to the x7that would be great

raphaelcoeffic, 2017-11-30:
@Kilrah I guess that you have the pins/port for these additional inputs? Maybe we could start by listing these additional inputs on each TX? Just so people know what to expect and what not?

kilrah, 2017-11-30:
I believe most should be already defined in the respective HAL files.

raphaelcoeffic, 2017-12-08:
@kilrah did I miss something? the only defs I can find are these:
```
#if defined(PCBX10)
  #define SWITCHES_GPIO_REG_GMBL        GPIOH->IDR
  #define SWITCHES_GPIO_PIN_GMBL        GPIO_Pin_14
  #define SWITCHES_GPIO_REG_GMBR        GPIOH->IDR
  #define SWITCHES_GPIO_PIN_GMBR        GPIO_Pin_15
#endif
```

That seems a bit odd to me, as I can see 5 pins on each side on the X10. Concerning the X7/X12S, I did not find anything.

rollysing, 2018-03-03:
Will this support the Q X7's P400 and P401 headers (top edge of the bottom board) for two additional ON/OFF toggle or push button just as [ErSky9](http://openrcforums.com/forum/viewtopic.php?f=5&t=6473#p90349) does? Hope it does.

sukru1968, 2018-04-04:
please can some one explain me how to eneble in opentx

3djc, 2019-06-24:
Done in 2.3

closed 2019-06-24 by 3djc

lshems, 2019-06-24:
Are they properly available in the getfieldid, and under which labels, if I may ask?

3djc, 2019-06-24:
si/sj for x7, gmbl/gmbr for horus

lshems, 2019-06-24:
Why not within the 'naming convention' for Horus? Any reason?

Just so I can take that into consideration using them. Different naming conventions might have functional reasons?

3djc, 2019-06-24:
On Horus, we have two (left and right) connectors, each allowing a combined 1 pot + 1 switch gimbal extension (https://www.horusrc.com/en/frsky-horus-x12s-stick-switch-with-pot.html)

lshems, 2019-06-24:
Still not clear to me. Sorry.

So the switches on Horus are named gmbl and gmbr and the pots have yet other names?

Sorry to ask, but is there already a Lua user reference guide published for 2.3? 

That would help a lot, as I understood it is autocreated from the coding comments.

Otherwise I need to check the code, fine. Just no clue yet where to look but I'll find it.

:)