opentx/opentx #5334: Limit checklist editor to available caracters

capitaineflam25, 2017-11-08:
I'm a french OpenTx user and accentuated caracters are common.
I don't need to display them on the radio but i regularly forget that they won't display well on it.

Maybe the checklist editor caracters could be limited to a caracter "whitelist" ?

[image]

[image]

kilrah, 2017-11-08:
You can actually write accented characters through their code (\200, \201 etc) but it' would be a mess to handle it in there.

capitaineflam25, 2017-11-08:
Thanks for the tips, i didn't know that.

PR #6434 (2019-05-20): Companion enhancements for checklist editor, model print and compare (2 3 branch)
Print and compare (includes PR #6349)
- Add global functions if enabled for model
- Add checklist contents if display enabled

Checklist
- Remove accents fixes #5334
- Display cursor position coords
- Force fixed width font
- Refactor to use newer version of removeAccents and move to helpers
closed 2019-05-20 by bsongis