opentx/opentx #5336: Recent files are duplicated in Companion

capitaineflam25, 2017-11-09:
This capture has been made with 2017-11-08 nightly build : 

[image]

kilrah, 2017-11-09:
Can't reproduce. Aren't those files in different folders?

capitaineflam25, 2017-11-10:
I found a method to reproduce it (Companion 2.2.1N431) :

- Create a new File
- Save it as "document1.otx"
- Create a model
- Save again
- Close Companion
- Open Companion : "document1.otx" should be duplicate in the recent menu

kilrah, 2017-11-10:
Nope no issue here with that procedure. Must be something related to your system...

capitaineflam25, 2017-11-10:
OK

For information here is what i found in my Windows 10 registry : 

[image]

capitaineflam25, 2017-12-07:
I've installed QT to help on another issue and now i can see what's happening here : 

This is due to QFileInfo that will uppercase drive letters (Source : http://doc.qt.io/qt-5/qfileinfo.html#canonicalFilePath)

If i replace **filename** with **curfile** on the 2 following lines, the issue disappear.

> files.removeAll(curFile);
> files.prepend(curFile);

@kilrah, can you reopen the issue or integrate the patch if you think it's safe ?

[image]

PR #5478 (2017-12-08): [Companion] Ensure no duplicates in recent files menu.
Fixes #5336

Thanks to @capitaineflam25 for thorough diagnosis!
commit 9c279473 (2017-12-08): [Companion] Ensure no duplicates in recent files menu (thanks capitaineflam25). Fixes #5336 (#5478)
closed 2017-12-08 by bsongis