opentx/opentx #5340: X10S and Crossfire, reciever stay on, but it's not.

Barbaryann, 2017-11-10:
Hello guys.

First you all do an amezing job.
Really thank you for all !!!!!

When i use a quad with crossfire, when i unplugged the battery on the quad. My X10S say it is already connected, and if i want to turn off my X10S, it say that the reciever is turn on. But it's not !

It works good with D16 protocol and X4RSB.

If think there is an issue with the crsf protocol on that.

schwabe, 2017-11-10:
Probablysome telmetry sensors are still sending even if rx is off. Do you get a telemetry lost message when you switch off the RX? You should get at least "Sensor lost"

If not what sensors are still shown as having telemetry (blinkiing * next to them) when the RX is off.

3djc, 2017-11-10:
It is checking specificaly for rssi

schwabe, 2017-11-10:
might be this
```
--- a/radio/src/telemetry/crossfire.cpp
+++ b/radio/src/telemetry/crossfire.cpp
@@ -153,7 +153,6 @@ void processCrossfireTelemetryFrame()
       break;
 
     case LINK_ID:
-      telemetryStreaming = TELEMETRY_TIMEOUT10ms;
       for (unsigned int i=0; i<=TX_SNR_INDEX; i++) {
         if (getCrossfireTelemetryValue<1>(3+i, value)) {
           if (i == TX_POWER_INDEX) {
@@ -163,6 +162,7 @@ void processCrossfireTelemetryFrame()
           processCrossfireTelemetryValue(i, value);
           if (i == RX_QUALITY_INDEX) {
             telemetryData.rssi.set(value);
+            telemetryStreaming = TELEMETRY_TIMEOUT10ms;
           }
         }
       }
```

schwabe, 2017-11-10:
I commited this als fix for this. Can you please retest in the next nightly if the problem is fixed?

closed 2017-11-10 by schwabe

Barbaryann, 2017-11-10:
Here a video on this issue :
https://www.youtube.com/watch?v=uOmmxwUaLhI

Barbaryann, 2017-11-10:
I test next day ;)
Fixed 40mn after the issue is posted.
Woooooo guy you rock !

schwabe, 2017-11-10:
Btw. while you are at it can you check the telemetry RX error counter under, long press menu on main screen -> statistics -> debug? (with 400k and 100k baud rate)

Barbaryann, 2017-11-10:
I check it, and confirm it is the same issue than the Q X7 (Shitty frsky !!!!!)
Works great in 100k and Many error in 400k

Barbaryann, 2017-11-10:
I can confirm you that this is this one we have to replace.
I am waiting the component to try this
[image]

Barbaryann, 2017-11-11:
Works well now.
Thanks you very much !

commit f9d5201b (2017-11-10): Only set telemetry to streaming when receiving rssi from RX on crossfire (closes #5340)