opentx/opentx #5342: Feature request: USB CDC to MSP over SPORT bridge

miradarya, 2017-11-10:
I'm dreaming of a way to connect a tablet or smart phone running a ground control station app like EZ-GUI  via USB OTG to the Taranis which then talks as a CDC device via MSP over SPORT to the flight controller. This would eliminate the need for the additional telemetry radio hardware when monitoring or controlling GPS navigation. Currently, MSP over SPORT is only possible from lua. Is it possible to configure the Taranis as a USB CDC device and passing the MSP data through that device?

schwabe, 2017-11-10:
USB CDC is already implemented for debugging puroposes with cli. Implementing something like a startSportMirror command should be possible.

kilrah, 2017-11-10:
Duplicate of #5209

closed 2017-11-10 by kilrah

miradarya, 2017-11-10:
Not really a duplicate. I am not talking about controlling TX channel values externally. I would like to tunnel serial data from UART CDC via UART over SPORT. That's an entirely different story, and imho even easier to implement.

kilrah, 2017-11-10:
Both are mixed together, as mentioned in that issue "and get telemetry" - "would open up a lot of cool scenarios such as UAVs and ground stations in general".

Telemetry is indeed the easiest and what would likely be done first, but the point is that we won't do one without planning for easy expansion to the other.