opentx/opentx #5348: X10s N33  Emergency mode on Power off

RoninMedia12, 2017-11-12:
When I do the long press to power off  it jumps in to emergency mode , If I hode down button again just cycles and powers off for a second then goes back into emergency mode, I must plug it into usb then hold down power so when it is off for that 1/2 it connects into stm32bootloader then it will stay off.   On frsky x10s

RoninMedia12, 2017-11-12:
So after doing some testing  i switched over to darkblue theme and issue went away.  But when i switch back to defualt it returns. I noticed the shutdown squence on the defualt theme is different the. The dark blue not sure if that matters.  I also reflashed n433 and it seemed to go away until i would change something on a model and would start again until i would go to dark blue theme hope this helps

3djc, 2017-11-12:
Duplicate #5270

closed 2017-11-12 by 3djc