opentx/opentx #5357: Option to mirror CRSF telemetry to the Serial port in the battery tray on Taranis X9D

oranzs, 2017-11-13:
Smartport telemetry can be mirrored without a problem. Is is possible to implement the mirroring CRSF protocol. I might be able to implement the functionality. Where should I start?

Amigache, 2018-05-28:
It would be interesting for those of us who use antenna tracker, and we use inav, for example.

ropalet, 2018-05-28:
I have a micro crossfire TX and I can see the telemetry from INAV in my taranis x9d, but it can't mirrored on Smartport telemetry port (serial). Can you implement this functinality?.

raul-ortega, 2018-05-28:
Hi, I'm also very interested on this implementation. I have realized that for FrSky telemetry is used this block of code to redirect the telemetry (mirrowing):

#if defined(SERIAL2)
  if (g_eeGeneral.serial2Mode == UART_MODE_TELEMETRY_MIRROR) {
    serial2Putc(data);
  }
#endif

Would this block of code solve the problem if it is included in src/telemetry/crossfire.cpp file?

Thanks.

fggy-x2, 2018-05-28:
I'm in the same situation with the X9D and the crossfire that I can't take the telemetry out,  to use it in the antenna tracker.

jgilbertfpv, 2018-06-30:
I'm having to forward my crossfire telemetry over bluetooth through a pc connection to my tracker right now, having crossfire support to do away with a pc in the mix would be great.

pixelscript, 2019-01-18:
For those with a crossfire micro that don't have the option of bluetooth bluetooth this would be very useful.

pixelscript, 2019-01-24:
FYI 3djc has put up a build here. I've can verify it's sending data on the TX line of the IO port but not been able to process it to confirm it's working 100%. https://github.com/opentx/opentx/commit/fcd8900cb33bfaff5b14da675dce519831865e93

pixelscript, 2019-01-24:
Yeah I noticed I had got best results at 115.2k baud. Decoding the signal proved to be beyond my skill set though, looks just like a stream of characters to me. Guessing it needs to be processed in a similar manner to `src/telemetry/crossfire.cpp`.

kilrah, 2019-01-24:
Yes of course

pixelscript, 2019-02-11:
FYI I've been using the changes from [here](https://github.com/opentx/opentx/compare/3djc/crossfire-telem-mirror), with it working well.

RomanFoltyn, 2019-05-14:
Hallo
I am very interested in the possibility of using CROSSFIRE CRSF to receive GPS Position and Altitude via the serial output x9D plus. I am just starting construction of u360 Tracker and it is extremely important to me.
Thanks. (Maybe there is a beta version with this feature?)
Roman
PS Greetings to Raul Ortega!

3djc, 2019-05-15:
I have merged the changes into 2.2 branch, which means that tomorows nightly will have that change. We are usually strongly advising against use of  nightlies. It this particular case, all change work is happening on 2.3 branch, so 2.2.4 nightlies should be safe to fly, but please make thourought test on the ground first

closed 2019-05-15 by 3djc

RomanFoltyn, 2019-05-15:
yesterday evening I uploaded 2.2.4 to my x9D Plus but this version does not send the signal through the port:
when the CRSF Telemetry works correctly in the S-Port Mittor Position, the port is constantly Low, Telemetry is constantly High, and the S-Bus position is always High.

But that was yesterday. Please enter what versions I should load

.I will wait until tomorrow and carry out the test again!
Thank you very much

RomanFoltyn, 2019-05-16:
Hallo 3jdc!

Unfortunately, the size of the compiled file is 0 Byte. 
I can not convince the computer to open it. ;)
best regards!
Roman DL6NBF

RomanFoltyn, 2019-05-16:
companion-windows-2.2.4N695.exe                    16-Apr-2019 05:30            13816876
companion-windows-2.2.4N696.exe                    16-May-2019 05:25                   0

3djc, 2019-05-16:
You can use yesterday companion to downloads todays firmware

shannonbaker, 2019-06-12:
What's needed to update taranis x9d+ to 2.2.4 to get crsf mirror available? I can't find a compiled 2.2.4. Do I need to figure out how to compile on my own or is there something pre-made?

3djc, 2019-06-12:
You need to get 2.2.4 nightlies. Get companion from here  https://downloads.open-tx.org/2.2/nightlies/companion/ and then the firmware relevant to your radio (obvisouly with crossfire enabled)

shannonbaker, 2019-06-12:
When you say 'and then the firmware relevant to your radio', is it that the companion software creates the firmware to install or do I need to get firmware from somewhere else?

3djc, 2019-06-12:
Your need in companion nightly linked above to select nightly firmwares (settings/settings/Application setting/Use nightly builds). You then need to setup you radio profile according to your radio and liking. You can then a firmware using file/download

shannonbaker, 2019-06-12:
Thanks. I' can't see anything related to CrossFire in the hardware serial screen on taranis after writing the new firmware and the 2.2 SD contents. I confirm that the nightly build splash screen Cobbs up when powering on the taranis. Is there something I'm missing?

3djc, 2019-06-12:
Have you set your serial port to 'S.Port mirror' ? (We need to change that name)

Saxin, 2019-06-17:
Everything is set (N2.2.4) and wired correctly - checked with R9 module, I'm wired to battery bay uart via inverter, set BT module to 115200,0,0 ...
Tried SmartPort telemetry viewer app nothing shows sadly - I'm not sure if it's the app or something else 
Any advise how this can be validated working ?

kilrah, 2019-06-17:
What app is it?

Saxin, 2019-06-17:
@kilrah 
This one - https://github.com/CrazyDude1994/android-taranis-smartport-telemetry, 
If you know some other way to validate it's working I would be very happy to hear, thanks.

kilrah, 2019-06-17:
Several things:
- This issue is about crossfire
- The S. Port mirror is used for internal, and may not be enabled with an external R9 since that's got its own plug you can connect to already.

Saxin, 2019-06-17:
@kilrah  

**" This issue is about crossfire"**
I'm using CF micro (R9 was used just to validate hardware is ok, inverter, BEC and BT module)

**"The S. Port mirror is used for internal, and may not be enabled with an external R9 since that's got its own plug you can connect to already"** 
I know I've connected to internal / battery bay UART and it works nicely with R9, I prefer the BT hidden inside battery bay.

Hope to somehow validate that crossfire sensors  / telemetry is being sent out ?

3djc, 2019-06-17:
This app looks like it is expecting smart port telem, does it know how to decode crossfire streams ?

kilrah, 2019-06-17:
What are you expecting to do with the crossfire stream? Just like for S-Port the crossfire stream is passed along as is, so you need an app that decodes crossfire stream. I don't know of any.

Saxin, 2019-06-17:
@3djc - I guess it doesn't, I was mistakenly under impression that internally in openTX all sensors are "converted" to smart port type and hence can be mirrored.

@kilrah  - I have x2 use cases, one is ground station display using some app able to decode CF tel. stream; second one is sending mostly GPS sensor to x2 trackers for video and control antennas.

putimir, 2019-08-19:
Hello all, I don't know where exactly to ask so: is this mirror possible to be directed to the "internal" BT module of the Horus family?

skaman82, 2020-02-14:
I am interested in this as well. I am using crossfire an would like to connect to an app via BT by mirroring Taranis telemetry via Serial Port @ 9600 baud. No app I have tried is working. Any tipps?

kilrah, 2020-02-14:
AFAIK there is no app that supports Crossfire telemetry.

pixelscript, 2020-02-15:
Thought I'd just leave this here. This is the web app I had started building a while back, it's no where near finished and might be in a broken state but if the code is of use to anyone have at it.

https://github.com/pixelscript/telemetry-reader

avs2, 2020-10-05:
Was this option removed? I cannot find it anywhere on my X9D Plus 2019 (non-SE) on 2.3.3. As far as I can tell no telemetry is output on the S-port pin in the battery bay.

3djc, 2020-10-05:
If I remember correctly, it is using a uart output, and none is available on 2019 version of x9d

avs2, 2020-10-05:
Ah, that would be a bummer. Was hoping to get crossfire telemetry into the Telemetry Viewer Android app.

avs2, 2020-10-05:
I may buy another transmitter to get this to work. Does the TX16S support this feature already?

3djc, 2020-10-05:
Yes, you can output this on one of the two bottom uarts

avs2, 2020-10-06:
Could I make a feature request to to support mirroring the telemetry to the Smartport on the 2019 edition? Or is this a hardware limitation?

3djc, 2020-10-06:
hardware limitation

avs2, 2020-10-06:
Okay. Thank for your help.

PR #6326 (2019-05-15): Mirror crossfire telem on serial
Output fixed at 115k (but fifo buffer is protected in case of overflow)

Users tested ok

This fixes #5357