opentx/opentx #5362: Unable to set a value greater than 999 with keyboard in Companion telemetry screen form

capitaineflam25, 2017-11-14:
If you want to set the chrono bar to 6000, you can do it with the +/- button but not by entering "6000" with the keyboard.

[image]

kilrah, 2017-11-14:
Radio type maybe? Version?

Can't reproduce.

capitaineflam25, 2017-11-14:
Companion 2.2.1N434 / Taranis X7

How i get the issue : 
- Create a new model
- Edit it
- Go to Telemetry tab
- Telemetry Screen 1 / Bars
- Select "Timer1
- Try to enter "1000" with the keyboard -> you should not be allowed to
- You can enter 999 and press "+" to enter something higher than 1000

kilrah, 2017-11-14:
Works here up to the allowed 1275.

[image]

capitaineflam25, 2017-11-14:
You're right, my mistake.

The "problem" (maybe it's not one) is elsewhere : 

On my radio i can setup a higher limit for the timer bar of "20:00" (which should be 6000 in raw value, as 1 unit (on Companion) is 20 ms of the timer)

So when reading back my radio settings to Companion, the value is truncated to 1275

capitaineflam25, 2017-11-14:
Should i open a new issue/enhancement for that ?

kilrah, 2017-11-14:
Just rename this one and clarify the first post.

PR #5437 (2017-12-19): [Companion] Adjust source value ranges and editors in LS/CF/Telemetry
Went down a rabbit hole with regards to timer and RTC ranges and how they're used/limited/edited in regards to LS (comparison functions), CF (set timerN action), and telemetry display bar limits.  Seems there's a bit of confusion here both in Companion and on radio (probably due to legacy issues).

Fixes #2984
Fixes #5362  (also negative values were not possible)
Fixes #5374
Fixes #5375
Fixes #5527
Fixes timer and RTC time ranges to comply with firmware limits.
Fixes not being able to use RTC time in LSs.
Fixes newly-added L [... cut]
closed 2017-12-19 by bsongis