opentx/opentx #5368: X10 + 2.2N433 / Blocks on debug statistics page

raphaelcoeffic, 2017-11-14:
While on the memory debug page (statistics), the Tx just blocked on the that screen, whereby the trim seemed to worked, as the haptic warning was present when trimming a bit left and then right. I could however not reproduce it. I had to unplug the battery.

I'll try to reproduce in the next days. I compiled with `-DPCB=X10 -DHELI=NO -DLUA=YES -DGVARS=YES`.

3djc, 2017-11-15:
Likely duplicate #5312, please test with N434 or later

raphaelcoeffic, 2017-11-15:
@3djc not going to change: I was already using HEAD of 2.2 branch, compiled yesterday, and I just checked that f6309138f6ae was already included.

kilrah, 2017-11-15:
More info needed, what setup (otx file?), any lua scripts or widgets used?

raphaelcoeffic, 2017-11-15:
No lua script involved, one widget (model picture with no picture configured) on main page. I'll upload my otx file ASAP.

raphaelcoeffic, 2017-11-15:
Tried to reproduce it, and it's not really easy, but I manage it ever couple of minutes while playing with that screen. What I noticed is: even if I don't get it to block, the "RTN" will not work every time. After pressing the wheel on the right a couple of time, it seems I need more presses on "RTN" before I can leave that screen. And sometime, it does not work anymore and I'm stuck on the screen.

[X10-files.zip](https://github.com/opentx/opentx/files/1475087/X10-files.zip)

raphaelcoeffic, 2017-11-18:
I might note that this might be related to the isues I had when using gcc 6.2.1. I'll have to retest to be sure it is somehow still happening.

raphaelcoeffic, 2017-11-19:
Rested with a firmware compiled on 2.2 HEAD with gnu-arm-gcc 4.7.4: **cannot reproduce anymore**.

BUT, I noticed that as soon as I used the rotary knob on the mem stats page, it takes at least TWO presses on [RTN] to exit the menu instead of ONE normally.

3djc, 2017-11-20:
@raphaelcoeffic could you try https://github.com/opentx/opentx/tree/3djc/Colorlcd-stats-menu-fix, I think it fixes the behaviour you mention

raphaelcoeffic, 2017-11-20:
@3djc thx, works for me in the simulator (could reproduce it there as well).

closed 2017-11-20 by raphaelcoeffic