opentx/opentx #5369: single call `crossfireTelemetryPush` blocks the telemetry output for smartport

raphaelcoeffic, 2017-11-15:
Steps to reproduce:
 - no crossfire module installed or configured
 - call `crossfireTelemetryPush` with a command parameter that is `!=` a polled sensor-ID.
 - then try to `sportTelemetryPush`, will never work.

raphaelcoeffic, 2017-11-15:
Same thing happens when:
- `sportTelemetryPush` is called with a sensorId not polled by the Tx-module.
- `crossfireTelemetryPush` is called with `command == 0`.

The solution is probably to offer some kind of `flush` function, so that we can force empty the buffer on script start or whenever it is necessary.

kilrah, 2017-11-15:
Or a timeout if no answer

raphaelcoeffic, 2017-11-15:
> Or a timeout if no answer

did you mean "on no answer timeout"? Well yes, that would be useful as well.

raphaelcoeffic, 2017-11-15:
Once we agree on the interface, I can probably implement it. I just need to know what is the preferred way to trigger that. Maybe just one special parameter to the same function? (in case a new function is a concern).

codecae, 2017-11-15:
Not sure I follow.  I'm as able to freely go between smartport and crsf models without blocking.

klutvott123, 2019-12-18:
@raphaelcoeffic A ``flush`` function that does ``outputTelemetryBuffer.reset()`` could be very useful. For example in the betaflight lua scripts where the telemetry protocol is detected by calling ``crossfireTelemetryPush/sportTelemetryPush`` it will fail if ``!outputTelemetryBuffer.isAvailable()``. For lua scripts that support multiple protocols it would be great to have a function like that to run at startup. Any chance that something like that could be implemented?