opentx/opentx #5372: Flexible storage model

raphaelcoeffic, 2017-11-15:
The idea here is to make the config storage flexible, so that new options can be added without having to convert everything.

Some requirements out of the top of my head:
 - should stay binary as the storage space is tight.
 - should allow new features without eeprom format change

Anything else?

raphaelcoeffic, 2017-11-15:
@tadango @kilrah @bsongis  please feel free to comment and add your ideas. The sooner we have a consensus, the sooner I/we can start implementing it.

RealTadango, 2017-11-15:
What is the difference between binary and JSON? It's all just a bunch of bits. A small size is important so don't store values matching default. JSON is great for that but long and repeated variable names take up a lot of space. I don't see a simple solution for that right away.....

bsongis, 2017-11-15:
I already have 2 branches with this work in mind, one with JSON, the other with YAML
All the libs I found use too many malloc and I would like to find time to rewrite something from scratch, without one malloc. This would allow us to use this storage method on Taranis as well.
Do you want to give me an hand with this work?

raphaelcoeffic, 2017-11-15:
@bsongis I can try to! To do it properly without malloc will basically mean calling some functions while parsing that will deal with the objects, so that you do not need to store anything. So basically some kind of streaming parser? I happen to use JSON a lot for my work, but the other one would be fine as well.

schwabe, 2017-11-15:
Or just generate a generate a parser that does not need memory since it directly sotres the information inside the structs.

raphaelcoeffic, 2017-11-16:
@schwabe you mean inside the memory structs for settings? Well, basically means the parser only streams the payload and calls function pointers instead of storing any data itself and passing them for later analysis.

3djc, 2017-11-16:
Unless most of the code is rewritten, the data needs to writte in the current data structs, have a look in https://github.com/opentx/opentx/blob/2.2/radio/src/datastructs.h

raphaelcoeffic, 2017-11-16:
Ok, so JSON or YAML? Please roll the fixes for me.

raphaelcoeffic, 2017-11-16:
I meant dices, of course ;-)

raphaelcoeffic, 2017-11-16:
I'd prefer JSON to be able to get rid of all these spaces and mandatory new lines. And I think json is easier to parse.

RealTadango, 2019-08-17:
Reply to keep it active ;)

raphaelcoeffic, 2019-08-17:
@RealTadango coming in 2.4 :-) Radio code for Horus already works, have a look at the PR.

RealTadango, 2020-04-08:
Keeping the issue alive :)

lshems, 2020-04-08:
Yep

raphaelcoeffic, 2021-05-14:
This is already merged into 2.4, and will be finished in EdgeTx.

closed 2021-05-14 by raphaelcoeffic