opentx/opentx #5375: Decimal entry is difficult in V2 / Logical Switches

capitaineflam25, 2017-11-16:
Companion 2.2.1N436 / Taranis X7

Language of the test computer is French (decimal separator is ",")

- Add a Volt telemetry sensor, name it A1, with precision 1
- Go to logical Switches and add a line
- Select "a<x", then "TELE1:A1"
- Select v2 default value so you can replace it
- Type 12,1 with the keyboard you should get "12,0 V"

There is no issue with other decimal fields, could it be linked to the presence of a unit ?

(This is a cosmetic issue, you can then select the value after the coma and replace it)

[image]

kilrah, 2017-11-19:
That's just the way the doublespinbox works in Qt.

capitaineflam25, 2017-11-19:
I don't know if the "Duration" field is based on the same doublespin box.
The issue is not reproduced on it.
I was wondering if the "bad" behaviour was due to the unit append at the end of the value.

[image]

kilrah, 2017-11-19:
Ah no, it's much more complex than that - since the values and steps that may be stored in an LS may not match what can be entered in the V2 box everytime that's edited we have to convert the value to what can be stored, then set the box back to the actual value. That overwrited what you just typed and cancels editing.

I don't see a way to improve that with reasonable complexity.

capitaineflam25, 2017-11-19:
Ok thanks for the explanation.
It maybe better to leave it as is.
I let you decide if you prefer leaving the suggestion open or closed.

closed 2017-12-19 by bsongis

capitaineflam25, 2017-12-20:
Thanks a lot, the new editor is way better !

commit cfbb30ce (2017-11-28): [Companion] LogicalSwitchesPanel: Update raw sources list upon screen refresh (fixes #5374); Improve data entry in spin boxes (fixes #5375).
PR #5437 (2017-12-19): [Companion] Adjust source value ranges and editors in LS/CF/Telemetry
Went down a rabbit hole with regards to timer and RTC ranges and how they're used/limited/edited in regards to LS (comparison functions), CF (set timerN action), and telemetry display bar limits.  Seems there's a bit of confusion here both in Companion and on radio (probably due to legacy issues).

Fixes #2984
Fixes #5362  (also negative values were not possible)
Fixes #5374
Fixes #5375
Fixes #5527
Fixes timer and RTC time ranges to comply with firmware limits.
Fixes not being able to use RTC time in LSs.
Fixes newly-added L [... cut]
commit 76ce186b (2017-12-19): [Companion] Adjust source value ranges and editors in LS/CF/Telemetry (#5437)

* [Companion] RawSourceRange: Fix for timers and RTC; Remove unused "single precision" flag and logic; Add logic for "short" timer handling.

* [Companion] Add TimerEdit class for editing negative times; Implement TimerEdit in LS, CF, and Telemetry (bars) editors;

* [Companion] Refactor LogicalSwitchesPanel for fewer & more efficient updates; Fix newly-added LS not being available in source lists; Fix bug with timer/rtc comparison values getting reset to value of spin box instead of time editor.

* Add forgotten #i [... cut]