opentx/opentx #5377: X10 crashing while switching between models (2.2 HEAD)

raphaelcoeffic, 2017-11-16:
These are the two first models ever created on this TX, and every time I click on "select model" with the other one, the TX crashes (screen gets black, power switch stays lit). Once I reboot, the model selected earlier has been effectively selected and runs. Trying to switch back to the first one, same thing again.

raphaelcoeffic, 2017-11-16:
Here are the otx file as well as radio.bin and both model bins:
[x10_issue_5377.zip](https://github.com/opentx/opentx/files/1480122/x10_issue_5377.zip)

raphaelcoeffic, 2017-11-16:
Firmware used:
[firmware.zip](https://github.com/opentx/opentx/files/1480211/firmware.zip)

raphaelcoeffic, 2017-11-16:
SD card content:
[x10_issue_5377_sdcard.zip](https://github.com/opentx/opentx/files/1480219/x10_issue_5377_sdcard.zip)

raphaelcoeffic, 2017-11-16:
code compiled myself, using HEAD of: https://github.com/opentx/opentx/pull/5367

Build options:
[build_options.txt](https://github.com/opentx/opentx/files/1480277/build_options.txt)

raphaelcoeffic, 2017-11-16:
Compilers used:
```
TARGET simu/libsimulator: cpp compiler /Library/Developer/CommandLineTools/usr/bin/c++ v4.2.1
TARGET firmware: cpp compiler arm-none-eabi-g++ v6.2.1
TARGET companion: cpp compiler /Library/Developer/CommandLineTools/usr/bin/c++ v4.2.1
```

kilrah, 2017-11-16:
Build environment issues as discussed in chat

closed 2017-11-16 by kilrah