opentx/opentx #5383: Horus X10 - Telemtry page for Lua script

Barbaryann, 2017-11-18:
Hello,
I don't find any thing to open a lua script in a telemetry screen like on the X9D+ for exemple.
The only way i find is to go into the SD card and execute manualy the Lua Script.
I am wrong  or not ?
And is it possible to make something for that ?
Thanks a lot for all your job to OpenTX team

kilrah, 2017-11-18:
There are no telemetry scripts on Horus, you have widgets instead.

Please keep usage questions on forum, not in the bug tracker.

closed 2017-11-18 by kilrah

Barbaryann, 2017-11-18:
I ask why there is not he same way than on the tarais, Because widget don't do PID for Betaflight...

kilrah, 2017-11-19:
Becasue the whole interface system is different. Someone can port the telemetry screen as a widget.

Barbaryann, 2017-11-19:
Is it possible ? It would be awesome.
where to ask this ?

kilrah, 2017-11-19:
You can try whoever wrote the script you'd like ported.

lshems, 2017-11-21:
Check my blog on rcgroups. Its shows the porting of an existing telemetry script to a widget. Not saying it is thé solution, but at least it gives an idea on a possible approach.

L shems