opentx/opentx #5393: [Horus] backup / restore on TX

raphaelcoeffic, 2017-11-20:
Similar to the "Backup EEPROM" function, it would be good to be able to backup relevant files on the Horus X10/X12 into a single package (on SD card) that could be restored just the same in a one step process (select backup file, restore radio & model settings).

I guess the corresponding Companion function could work just the same and use a compatible format, so that backup files can be used either on companion or Horus (use OTX file?).

3djc, 2017-11-21:
It could be done in a possibly simpler way by adding a date tag to the two files and keeping them in the dir ?

raphaelcoeffic, 2017-11-21:
Right, that would work. But then it’s a pretty mess to backup these files by hand somewhere else (like copy them to your hdd) or open them directly in companion.

kilrah, 2017-11-21:
If you find a lightweight open source zip lib, no prob...

raphaelcoeffic, 2017-11-21:
Is OTX actually zipped? Concerning the bootloader port, I think I will start with cleaning up the platform specific code, so that it gets easier (found `loadboot.cpp`, which is everything but platform independent.... the bootloader itself has some rest as well).

projectkk2glider, 2017-11-21:
Yes otx is normal zip archive.

raphaelcoeffic, 2017-11-21:
Is it using `DEFLATE`? or some other algo? Just seen that it is actually possible to store files in a ZIP file without actually compressing them. So theoretically, a compliant ZIP file could be created without much overhead. However, some kind of decompression should be supported, so that OTX files created by companion can be restored, right?

raphaelcoeffic, 2017-11-21:
@kilrah @projectkk2glider seems `miniz` is already used in companion. Did anyone try how much memory / rom it would cost to use this little gem?

schwabe, 2017-11-21:
I think space on the sdcare is not really a problem imo.

raphaelcoeffic, 2017-11-21:
@schwabe the idea is not to save space, but use OTX files for backup/restore, because it is directly usable in Companion. To open backups from Horus in Companion, it is probably sufficient to implement the "no compression" file format. But I guess that restoring files coming from Companion on the Horus requires support for `deflate` which is what is used in OTX files (just checked). It would be a pitty to use yet-another-format, whereby there is already one.

raphaelcoeffic, 2017-11-21:
ok, so let'a make it simple:

*model backup / restore:*
- use simple copies of the models with `filename = [model name]-[date-time].bin`
- copies are stored in `/MODELS/BACKUP/`, so that it is not confused with the running models, and the restore pop-up just has to list everything in that directory that has a `.bin` extension.
- companion should be able to open these files without any changes, right?

*full tx backup / restore:*
- maybe support OTX if `miniz` does not use too much memory
- implement that later

projectkk2glider, 2017-11-21:
Horus does not have any serious memory limitations, Lua is using 8MB DRAM. I recommend you look into using existing minz lib from companion in radio firmware source for this.

raphaelcoeffic, 2017-11-21:
Do you mean for the full backup, or for single model?

kilrah, 2017-11-21:
Full of course. I don't think we'll do single model,. we have nothing to handle those anywhere.

lshems, 2017-11-21:
I would love a single model copy also for Horus. Perhaps I'm missing something? It is very convenient in the field being able to copy just a backup model from one transmitter to the other and restore it there.

raphaelcoeffic, 2017-11-21:
Yes, I've heard from a friend he used the single backup model on the Taranis a lot to avoid always copying the model before applying modifications.

kilrah, 2017-11-21:
> the next week-end, you don't remember which of the models was the backup, and which was the modified one

That's his fault for not renaming the copy though, seems obvious enough...

raphaelcoeffic, 2017-11-21:
@kilrah for sure, but that's the way we are: sloppy and slowly getting old.... (at least I do)

raphaelcoeffic, 2017-12-10:
@projectkk2glider after many things done/tried, the only way miniz works in the horus is when I use 0-compression. Otherwise, I get all sorts of wild memory corruptions, extending to SD card corruptions, hard faults, etc... It's quite weird if you ask me. When executed in the simulator, everything works smooth.

projectkk2glider, 2017-12-11:
Are you doing this in the bootloader or not. If yes, then you should check the linker file where the mallloc memory regoion is defined.

raphaelcoeffic, 2017-12-11:
@projectkk2glider nope, it's in the normal firmware.

sly1231, 2018-09-29:
How does one backup models on a Horus X10 (OTX) ?
Thanks

3djc, 2018-09-29:
You just backup de model files from your SD card

sly1231, 2018-09-29:
Thanks 3djc

sly1231, 2018-09-30:
How does one use TEXT widget in open TX

Luznatural, 2019-08-23:
Any progress in this issue?
Thanks in advance

sly1231, 2019-08-24:
I read somewhere that you could create a text file and add it to the model file and it would come up as the opening screen.  Have not had the time to try it.

closed 2020-05-20 by stale[bot]