opentx/opentx #5394: Phantom firmware for the Taranis Q X7S

jakezilla, 2017-11-21:
I can't find firmware for the Q X7S anywhere, and if I try to flash the X7 firmware it gives me a not compatible error. Since it shipped with what appears to be the latest firmware, are we just waiting till the next OpenTX to fully support the radio?

On the radio-
FW: opentx-x7s
VERS: 2.2.0 (9ad5a5be)
DATE: 2017-09-04 20:37:35
EEPR: 218

3djc, 2017-11-21:
What you have installed was a shapshot of a work in progress support for X7S.

In the coming (very soon (tm)) 2.2.1, X7 and X7S will run the same firmware.  You may get the 'not compatible' error the first time you flash from your current firmware to the new type, but not afterward

closed 2017-11-21 by 3djc

schwabe, 2017-11-21:
@3djc aka the firmware that frsky preloaded. We may need to teach Companion that X7S and X7 are the same when flashing X7