opentx/opentx #5400: Feature request: easy servo setup

patman13, 2017-11-21:
I'm coming from Multiplex. Maybe you could put this on the feature wish list.

When modifing Servo values (min, max, center) the values are sent to the receiver. So no extra input at the sticks are necessary to set up the servos. Hands free for set up the mechanic (e.g. plane: rudder linkage).

patman13, 2017-11-29:
one more detail:
the value you are editing (the blinking one) shall be written to output

kilrah, 2017-11-29:
There needs a way to enable that, like holding the MENU key during editing or something for safety... but yes the idea is nice.

RCdiy, 2017-12-24:
Example to consider. 
At 1000us the servo is well past the binding point. 
Moving sticks we can see the problem and stop. 
If we edit lower limit which defaults to -512 then we have severe binding immediately on editing that value. 

Good idea but make it optional with default off.

kilrah, 2017-12-24:
That's why I said what I said. 
Set limit back to like 50% first, and only then hold the "enable" button and increase value to find the right one.