opentx/opentx #5401: Widgets

radlouni, 2017-11-22:
i just flashed the Nightly 440 and noticed that some widgets dont work, for example if you specify in widget settings that the sensor is RxBT or RSSI, the widget is blank. However, the "Cels" sensor works and displays the total battery voltage and individual cell voltages

kilrah, 2017-11-22:
Radio type ?

radlouni, 2017-11-22:
Frsky Horus X10S

Thanks

3djc, 2017-11-22:
Have you ensured those sensors have a proper value in telem screen ?

radlouni, 2017-11-22:
Yes, RxBt and RSSI had proper values on telem screen, the receiver was
powered on and bound with the hours x10s

Thanks

radlouni, 2017-11-23:
i also flashed the 441, and still the same issue. I checked again to make sure that the sensors have values in telemetry screen, and i created a widget for the FLVSS sensors (Cels), and the widget displays FLVSS sensor data properly, if i edit the same widget and i chose a different sensor (RxBT or RSSI), then the widget is blank. if i try to create a new widget using RSSI or RxBT sensors as the source, the widget is blank.

radlouni, 2017-11-23:
I am closing this. The widgets work, I just did not know how to use them

closed 2017-11-23 by radlouni