opentx/opentx #5402: Companion 2.2.1 build 440: UI bug during eeprom translation from X9D to QX7

RC-SOAR, 2017-11-22:
The problem arises in Companion 2.2.1 (latest nightly build) when (a) translating a model from X9D to QX7 format AND (b) editing a mix with a source which is not supported by the QX7 (e.g. 'LS'). The EEPROM format of the source data is 2.2.

- If you open the mixer editor on and change the source from 'LS' to a valid control for the QX7, there's no problem.
- However, if you open the mixer editor on the LS line and then press OK without changing to a valid source, all subsequent mixers are deleted.

Minimalist .otx file is available [here](https://www.dropbox.com/s/3rj7bhe9um4jdgw/UI_Bug.otx?dl=0).

bsongis, 2017-11-22:
I will take it.

bsongis, 2017-11-22:
Reproduced. Thanks for the good explanation + the minimalist .otx, half work is already done!

closed 2017-11-22 by bsongis

RC-SOAR, 2017-11-23:
Many thanks :-)

commit d6494202 (2017-11-22): Fixes #5402 (#5405)