opentx/opentx #5412: X9E Operation Time Reset

bosshog99, 2017-11-23:
Hello,

I'm running version 2.2.0 on a Taranis X9E and no longer have the option to reset the timer on the color display.  I like to reset this timer each time my transmitter battery is recharged to track it's performance.

thanks,

3djc, 2017-11-23:
X9E ? Color display ?

kilrah, 2017-11-23:
The top display. Long MENU on statistics page, still there.

closed 2017-11-23 by kilrah

bosshog99, 2017-11-23:
Yes, the color display.  it is there but, what i was asking, I cannot reset the operation time.  this is the timer on the color display. before 2.2.0 this timer was reset in the status menu.  now the timer shows up in the status menu but the timer cannot be reset.

kilrah, 2017-11-23:
Don't know about 2.2.0 but tested working on the soon to be released 2.2.1 so update to that once it's out.

3djc, 2017-11-23:
Yes, I fixed that a while back