opentx/opentx #5420: TBS Screen has a flashing white line going across the middle of the screen

Crunch69, 2017-11-24:
x10 N442 when you switch to TBS screen there is a flashing white line going across the middle of the screen

kilrah, 2017-11-24:
What do you refer as "TBS Screen"...?

Crunch69, 2017-11-24:
TBS screen = TBS Theme

kilrah, 2017-11-24:
OK. The crossfire configuration script could also be a "TBS screen".

loetefix, 2017-12-06:
On my X10S the same but also the Inputs of the sticks make weird thinks with that theme....the Channel output jumps totaly wild.

offer-shmuely, 2018-09-11:
To fix the TBS logo and place it on the correct palace (currently it can not be seen)

on THEMES/TBS/main.lua:36
replace the 50,50 with 0,0
i.e.
  lcd.drawBitmap(bitmap, 0, 0)

fpinto14, 2019-06-17:
Same problem.  The fix above seemed to work but on reboot I got the emergency mode screen.  Had to restore backup SD to get back to booting.  Still have white line in TBS theme.  :(.