opentx/opentx #5422: Off brightness slider has no effect on Frsky X10S

putimir, 2017-11-24:
I noticed this issue quite some time builds ago, but didn't put much thought into it, I thought it was may a temporary "glitch", but it persists, and since the 2.2.1 milestone is around the corner, I thought I should report this.

So as the subject states, the "off brightness" seems to have no effect on the actual brightness; I remember that in the versions around 40x, if I set the slider all the way to the left, the display had almost zero backlight, but now I think there is no difference between on and off brightness... also I think that the brightness range is much smaller for the on brightness (in a way you could say that brightness cannot be lowered to zero in either mode)...

Does anyone else have this problem?

raphaelcoeffic, 2017-11-24:
@putimir I think that's more or less engineered this way. I've been told that there were display glitches when the brightness would go bellow a certain value, which has now been deemed as the minimum allowed.

putimir, 2017-11-24:
Ok, but the fact is, that the setting was working beautifully on earlier versions - it did dim out almost completely, at least on my X10s; my question is why was this removed...

raphaelcoeffic, 2017-11-24:
@putimir it wasn't me ;-) no clue what's behind that decision, really. I heard there were some gliches on the screen or something like that....

putimir, 2017-11-24:
👍 

Ok, too bad, since it's most waste of battery juice hehe. Thanks and sorry for opening this, I could have researched more beforehand...

closed 2017-11-24 by putimir

trojanGoat, 2017-11-24:
@putimir,  let me know if you end up finding a solution for this.  I was also inquiring about the same issue.  The frOS seems to be able to do it.

s700ky, 2018-02-05:
You cannot set brightness levels in GV for backlight to a pot