opentx/opentx #5423: Hide "_." UNIX files

reptil1990, 2017-11-24:
Hey team,

Mac is creating files with "_." prefix. Macintosh is not displaying these files in the explorer, but they show up in the OpenTX SD card browser.

This will become a real big problem if an inexperienced user selects such a file to flash a new firmware to an RX. This will brick the receiver and it is not usable anymore.

Is it possible to hide this files? They have no use at all!
Thank you for your work!

patman13, 2017-11-25:
try "CleanMyDrive2"! that solves your problem.

reptil1990, 2017-11-25:
It is not a problem for me I know the deal and not to select such a file but there are many users don’t know such a tool!

schwabe, 2017-11-25:
What version of OpenTX are you running? 2.2.1 will already hide hidden files and unix dotfiles

closed 2017-11-26 by 3djc