opentx/opentx #5424: simulator on Companion not working (N443/N444)

x10user, 2017-11-25:
TX simulator does not start with an error message (since N443/N444) on the first attempt but starts at the second attempt.

TX to simulate: X10

Worked properly with nightly builds before.

[image]

kilrah, 2017-11-25:
Which of the 3 different methods are you using to start the simulator?

x10user, 2017-11-26:
If started from the program main window and started from the model edit window it gives an error message. Not tested if the simulator is started from shell.

mpaperno, 2017-12-18:
@x10user Can you still reproduce this in 2.2.1?  If so, please provide a .otx file to test with and re-open the issue.

closed 2017-12-18 by mpaperno

Shonkonray, 2018-01-29:
Hello,

I am new to OpenTx and I think this is still an issue, because i get the same error, or i havent found the soluten yet.
So i didn´t wnat to open a new issue, but cant reopen somebody elses.

I noticed a further problem with the stand alone simulator.  I wanted to use the .otx file to configure my display and download it to the TX (i did it once and it worked like a charm)

My usual way: Open the standalone with otx file and edit my widgets. Close the sim and be happy.
[image]

But now with a click on Ok; First Error:
[image]

With the click on Ok again; Second Error:
[image]

My .otx File incl. Log-File of the Error:
[Horus_X10S.zip](https://github.com/opentx/opentx/files/1674040/Horus_X10S.zip)

I have to mention it worked for me one day without any flaws and then suddenly it was broken. As i remember the only thing i did was updating Teamspeak and Steam.

What i did so far:
1) Reinstalled everything and deleted the registry
2) On my first attempt of ever using OpenTx i tried a nigthly build, but this had some troubles with simulation the display correct from the very beginning. After this i switched to 2.2.1 and it worked, as mentioned above, for 1 day.

Setup:
Windows 7
Companion Version 2.2.1; Dec 17 2017

I hope you can give me some advise, or at least my files are somewhat useful for you.

Thanks.

3djc, 2018-01-29:
[E] SimulatorWidget::useTempDataPath():349 - ERROR : Failed to create temporary settings directory ""

There may be something wrong with your TMP/TEMP dir seetings on Windows

Shonkonray, 2018-01-29:
Thanks for your response. Something wrong, just dont get to a point where i can test something out, do have a path where a temp dir should be?

3djc, 2018-01-29:
In your pictuire above, the data folder path is empty

Shonkonray, 2018-01-29:
I tried giving it a folder there, but the issue is the same. Also i wouldnt understand why i have to selct all three paths if i only can select one option to use.

mpaperno, 2018-01-29:
You do not need to enter a Data Folder, but you do need a valid TEMP path, as @3djc mentioned.  What do `TMP` and/or `TEMP` evaluate to on your system?  You can check quickly by using a command prompt with:

`> echo %TMP% && echo %TEMP%`

Should show something like "C:\Users\username\Local\Temp", twice.  Next step would be to verify that you have permissions to write files in there.  Another possibility is some overactive virus scanner preventing temp file creation.

Unfortunately the logged error message from Simulator doesn't indicate the actual temp directory being tried, instead it improperly shows the radio data path folder (which in this case is blank).  That needs fixing, but it's a minor issue.

mpaperno, 2018-01-30:
BTW, as a workaround you could just copy the SD card contents to a local folder and use that as the startup option.  Then copy models/settings back to radio SD when done (you can even do individual models and/or the radio settings separately this way). 

Also, a `.otx` file is just a Zip-compressed copy of the RADIO and MODEL folders as found on the SD card... if you rename it to eg. "x10.otx.zip" then you can use standard Windows de/compression tools on it (or add an "Open with..." option to context menu for .otx files).  Put the data into a temporary directory and point the simulator Data Folder to that location. When done you can re-zip the data into a new .otx file.  This way it could also be used in Companion directly.  This is basically what the simulator is doing for you when you start up with a .otx file.