opentx/opentx #5425: Companion crashes when try to edit radio settings (N443/N444)

x10user, 2017-11-25:
since N443/N444 Companion crashes when try to edit radio settings (for X10)

This worked fine in older nightly builds.

[image]

kilrah, 2017-11-25:
Details please... blank model? Already some data? If yes supply the file

x10user, 2017-11-25:
Observed with active TX-Type Horus X-10 but I can reproduce it also with X9D+.

Further information (don't know if valuable): Windows 10, Companion application language is English.
What kind of file helps?

3djc, 2017-11-26:
We need a reliable way to replicate the problem

kilrah, 2017-11-26:
The otx file that causes the crash if any, or instructions to reproduce the problem from a blank configuration.

x10user, 2017-11-26:
[FrSky X10.zip](https://github.com/opentx/opentx/files/1504178/FrSky.X10.zip)
I attached the otx-file for reference.

It does not crash if I start from an empty otx-file.
I was wondering if it might been caused by the assignment of trim buttons to the adjust GV function from the special functions in the TX settings. These are enabled (at least were enaled in earlier nightly builds) but not available at the system menu at the transmitter. This is an inconsistency between Companion and transmitter firmware.
Other than that I set up some telemetry screens and sensors at the Transmitter whereas these options are currently not available at Companion.

closed 2017-11-26 by x10user

kilrah, 2017-11-26:
Most likely. Adjust GVAR is not possible in global functions, removed now.

commit 1c1624b8 (2017-11-26): [companion] Adjust GVAR not possible in global functions (fix #5425)