opentx/opentx #5429: Crossfire heading missing

rFlightFPV, 2017-11-26:
TBS Crossfire telemetry has no field for heading if I write a logfile (SD logs). According to TBS the heading data is included in the GPS frame. Would it be possible to parse that value and put it in the "Yaw(rad)" field, which currently does not populate (always zero)? Futhermore, if this is under OpenTx control, I'd like to have the heading as a "sensor" as well so it can be displayed on the radio telemetry screen.

kilrah, 2017-11-26:
I believe it should work as is. What's the source of the data? Most likely that does not supply the heading to the receiver.

rFlightFPV, 2017-11-26:
@kilrah Datasource as said TBS Crossfire telemetry. According to Team Blacksheep the heading data is sent with the GPS frame. However there is no field named heading in the SD log. And the only existing field where heading would make sense is "Yaw(rad)" which is not populated.

kilrah, 2017-11-27:
What I'm asking is what kind of device is providing GPS data to the receiver to send down. Does it come from a GPS module connected to the receiver (which one?), or is it a flight controller (which one, which software is it running etc?)
Yes the protocol supports heading in the frame and there is code in OpenTX to handle it, so the device onboard may not be populating the value.

rFlightFPV, 2017-11-27:
The device sending the data is the Team Blacksheep (TBS) GPS/Compass which is connected to a TBS FPVision stack (with TBS Colibri FC) via BST (BlackSheep Telemetry) cable. I do get the heading on the OSD, so the unit is sending the data to the FC. According to TBS the data should be passed on in the GPS frame. There is a GPS field in the log (and available as "Sensor") which contains latitude and longitude separated by a space. No heading information anywhere. (Splitting latitude and longitude into two separate fields would also be nice.)

rFlightFPV, 2017-11-27:
I re-checked my latest DVR recordings and the heading in the OSD works perfectly after arming. Just to be sure I ran the sensor discovery while armed (with the heading on the OSD) and there definitely is no sensor regarding heading. I also wrote an SD log during the test, no heading field and "Yaw(rad)" is always zero. A fix would be greatly appreciated, thank you in advance.

tgreer, 2017-12-31:
I've noticed this also, running inav and a ublox GPS however same issue, heading or ground speed doesn't appear to be being detected as a sensor.

tgreer, 2017-12-31:
Just for clarity my setup is:
Inav v1.8.0 running on an Omnibus F4 Pro V3 talking crsf to a micro rx
Micro TX in Taranis X9D running Open TX 2.2.1

tgreer, 2017-12-31:
https://github.com/opentx/opentx/blob/2.2/radio/src/telemetry/crossfire.cpp#L72-L77 
This shows the fields in L74/L75, however the sensors aren't auto detected, and manually setting them doesn't make them be populated.

tgreer, 2018-01-01:
@bsongis As you did the initial implementation, do you have any input on this at all?

tgreer, 2018-01-02:
Seems there are some CRC errors. 
00 03 ea 0c 14 [CRC error]

3djc, 2018-01-08:
I have refreshed the python crossfire parser to help debug this, but all logs that we have been sent have wrong data for heading (0xff 0xff)

tgreer, 2018-02-25:
This should be fixed in a new version of crossfire firmware @rFlightFPV. I was working with Remo to sort and test.

3djc, 2018-02-25:
Thanks for sharing, as I could not find the issue on our end. let me know when it test ok with the new firmware and I will close this

tgreer, 2018-06-02:
@3djc I've tested this and it's working now. You should close this off.

3djc, 2018-06-02:
👍

closed 2018-06-02 by 3djc