opentx/opentx #5434: Feature Request: Custom Splash Screen for Horus X10

trojanGoat, 2017-11-27:
I don't see options for using custom splash screens for Horus X10 when powering up the radio.

https://imgur.com/a/vJYeH

raphaelcoeffic, 2017-11-28:
Yep, I'd really prefer to have custom splash images being just stored on the SD card, so the process is way easier:
- no need to patch the firmware on the fly
- automatically there when flashing a new version.

commonspace, 2018-03-18:
Bump. 
Is this feature going to be added?

3djc, 2018-03-18:
Will be in 2.2.2

closed 2018-03-18 by 3djc

R3M1X972, 2018-08-14:
Hi guys, anyone got news on this feature ? I still can see the "edit splash screen" button when connecting my Horus X12S to Companion 2.2.2 freshly updated, but the button is grey and can't be used, any ideas ?

3djc, 2018-08-14:
maybe you could read 2.2.2 release notes

R3M1X972, 2018-08-14:
@raphaelcoeffic & @3djc : thanks guys. However I did created a splash.png and placed it in the SD as suggested in release notes (/IMAGES/splash.png). I can see the file displaying when navigating through the menu but for some reason it doesn't shows up on Horus start (OpenTx screen is still showing up instead)... did I miss something again ?

raphaelcoeffic, 2018-08-14:
@R3M1X972 you are using 2.2.2, right?

R3M1X972, 2018-08-14:
@raphaelcoeffic yes, indeed, I'm on 2.2.2. could png image size be the solution?

kilrah, 2018-08-14:
It must be the the right size for the screen or smaller.

raphaelcoeffic, 2018-08-14:
@kilrah in fact no, it will work with bigger images, will just get cropped.

R3M1X972, 2018-08-14:
Dammit, I'm sure i'm missing something but can't figure what it is. Created the splash.png in 3 different sizes and placed it subsequently in the above mentionned folder, but still I'm getting the "opentx" logo when powering up the Horus :-/ 
Should you have any idea it'll be really appreciated ! 👍

raphaelcoeffic, 2018-08-14:
@R3M1X972 could you please come to our chat (chat.open-tx.org) and post me your image?

R3M1X972, 2018-08-14:
Hi all - my issue is now solved thanks to the help received from @raphaelcoeffic & @kilrah ! thanks guys.

splenkingforth, 2018-08-30:
Hi 
Could you please share the solution that you found?
Many thanks

R3M1X972, 2018-08-30:
Hi @splenkingforth
Solution found was to create a png named "splash.png" into the "image" folder of the SD card and redo the update of firmware. However for some reason on my Horus, the custom splash only works with a 480 × 272 file, meaning mine seems unable to resize a bigger image.