opentx/opentx #5436: [Companion] deleting a mix on an imported model kills the complete model

raphaelcoeffic, 2017-11-28:
Here is a file containing an easyglider model which I copy pasted from my X9D+ model file converted to X10. If you try to delete mixes on channel 15/16, the complete model gets wiped off / removed from the file.

The funny thing is: if I use `backspace`, I get asked whether or not I want to delete that model. Then I cancel, and use `right-click` and select `delete` on the CH15 mix. Then I get asked whether or not I want to delete that mix, which I confirm, and then the model is just gone.

[X10_models-26-11-2017.otx.zip](https://github.com/opentx/opentx/files/1510973/X10_models-26-11-2017.otx.zip)
Please note that I had to rename the file for github to accept it. JUst rename it back to try it out.

mpaperno, 2017-11-28:
Can't repro this...  😕  Mix lines are properly deleted with key or context menu.  Am I supposed to see the model disappear from the list right away, or only after saving/re-loading? Using CPN on Win7 built from latest 2.2 branch, MSVC2015, Qt 5.7.1.  

Also backspace key does nothing, but delete key acts correctly (asks to delete selected mix line). It would ask to delete model if focus is in file window... can't think of another reason.  Funny peculiar!

raphaelcoeffic, 2017-11-28:
@mpaperno maybe this is a focus thing with OSX.... Here it fails 100%. I might try to debug this on my side.

mpaperno, 2017-11-29:
On OS X I can recreate the focus issue with key commands, but not the model getting deleted.

It's like the popup model (or radio settings) editor dialog doesn't get full focus after opening (or being switched-to after using main window).  All key commands are still being caught by the main window (not just delete but copy/paste, etc).  If you edit any model field and trigger the "file changed" indicator, then you get full focus in the editor window and keyboard shortcuts are correctly caught there.  You can tell when this happens because the highlight disappears from the selected model in the main window.

[image]

[image]

I have no idea what the solution is... might try a few things but just shots in the dark.  I can't get the model to be deleted like you described though.

If you save the file with the "deleted" model (eg. as a new file), then re-open that version, is the deleted model still gone?  We've seen phantom disappearing Horus models before, though that was a while ago.

raphaelcoeffic, 2017-11-29:
Don't ask how that happened, but now I cannot reproduce the "model deleted" issue anymore... The focus thing works now exactly like you described earlier.

mpaperno, 2017-12-07:
Were you able to repro the deletion issue, or should we change the title?

I tried a few things for the focus issue but haven't found the solution.  The crux of the matter seems to be the QTreeView (in MdiChild) keeping focus when a line is selected.  Once the file is modified, the list is refreshed and the current model is no longer selected (which is also a bug), then the focus shifts to the currently active window.

raphaelcoeffic, 2017-12-07:
I guess the title can be changed, indeed. Thx for the explanation, it's getting really funny!

raphaelcoeffic, 2019-08-27:
works fine on 2.3, apart from the selected model that is not selected anymore once the model changes, but that's alright, nothing annoying.

closed 2019-08-27 by raphaelcoeffic