opentx/opentx #5438: Nightly Build does not work with X10 and SXR receivers

ftherrmann, 2017-11-29:
I've bound a S8R and S6R using new model setups in their default state using two servos for Ail (c1 and c5) and motor on channel 3.  Both receivers move the channel 1 Ail, but no rudder movement and only a slight amount of movement in one direction for the elevator.

ftherrmann, 2017-11-29:
This was a nightly from yesterday.  The throttle also works.

kilrah, 2017-11-29:
Check the channel monitor moves as expected... but I can't think of anything but a configuration mistake either in your model setup or the receiver itself.

ftherrmann, 2017-11-29:
Everything on the monitor works correctly.  Do you understand what information is passed to and from the SXR receiver and transmitter when a reset of the receiver occurs?  i.e. holding the bind switch or toggling the channel 12 output?  I know it has something to do with the end limits of the servos and is apparently stored in the receiver.  I will revisit tomorrow.

3djc, 2017-11-29:
Did you properly conduct ch12 servo min/max learning ? SxR are wonderfull, but require proper setting and calibration

closed 2017-11-29 by 3djc

ftherrmann, 2017-11-29:
3djc,   Has OpenTx been tested with the X10 and the SXR?  I would assume it has.

3djc, 2017-11-29:
As far as OpenTx is concerned, it doesn't care (or even know) about your receiver exact type. All it knows is it D16; like X6R and all the rest

ftherrmann, 2017-11-29:
That's what I thought.  However I do know that information regarding servo end limits is stored in the SXR, unlike the X series receivers.  I wish I could find a reference which explains this.

3djc, 2017-11-29:
In the frsky user manual, thats the section linked to ch12.

ftherrmann, 2017-11-29:
By user manual you're talking about the one page instruction.  Based on memory it only talks about a receiver reset by toggling ch12, but says nothing about what gets reset.  I know that the configuration does not get reset.  I'd just like to know more about what information is transferred from the Tx to the Rx during a ch12 reset.  It seems that this is a big mystery for some reason.

3djc, 2017-11-29:
Please download and read the manual, there is a whole parapgraph on that:

[image]

ftherrmann, 2017-11-29:
Thanks.  It was working fine under FrOs but apparently something broke during the flashing of the nightly.  I'll check it out tomorrow.

kilrah, 2017-11-29:
There is a full manual several pages long for the SxR on FrSky's website. You likely initiated limits calibration but didn't move the sticks to set them afterwards. This is not an OpenTX thing.

>Do you understand what information is passed to and from the SXR receiver and transmitter when a reset of the receiver occurs?

There is none. You just have to follow procedure and do the right things the SxR wants you to.

mpjf01, 2017-11-29:
What you are observing is typical of what happens when you don't do the self check either correctly or at all. The procedure is described in the leaflet that comes with every receiver. Adding to what is there I suggest that you do not move the throttle stick when setting the limits for the aileron, rudder and elevator, only after these controls have been moved to their limits should the throttle control be moved.

ftherrmann, 2017-11-29:
I just wanted to say thank you for the information.  The SXR is now working with the nightly release.  However I would like to caution that the operation especially the calibration (CH12) of the SXR can unreliable and inconsistent at best.  I've played with it for awhile this morning and here are some of the weird things I've seen:

1.  Calibration will "sometimes" reset the mounting configuration back to normal/flat.
2.  When switching from modes such as knife edge or hover I've seen the SXR drive the ailerons incorrectly.  Unplugging the receiver and powering back up fixes the problem.
3.  For some modes the gain setting actually controls gain for ail and rudder.  I've read incorrectly that gain ONLY works in gyro normal mode.  This is apparently not the case.
4.  Moving rud, ail and elv at the same time during calibration seems to confuse the SXR "sometimes".  As best I can tell the best method of calibration is to move one control at a time during the blue light on calibration.
5.  And finally...  Check, recheck and double recheck that all modes work correctly before flying!  I personally don't trust this receiver enough to use it in an expensive plane.  Until the firmware (I'm using the latest) logic becomes more robust and the calibration procedure and operation are clarified I'm only using it in a 3D EPP type park flyer.

3djc, 2017-11-29:
Sure, but you do realise that we have no link with frsky, right ? It is probably better to share your findings on frsky forum, or rc enthusiasts site like rc group

ftherrmann, 2017-11-29:
It's not your problem.  I was just sharing my experience and the fact that the flaky nature of the SXR setup and operation can make for some interesting experiences.  It can get really confusing to the point where your typical user could easily mistake the problem for an OpenTx issue.  Especially for that first time user.  I've used a couple of the SXRs before with FrOS.  I have already posted it on the FB X10 forum (with the note that it's NOT a OpenTx problem) and I will also share this with FrSky.