opentx/opentx #5452: [Feature request] Option to boot directly to certain default model

emski0, 2017-12-01:
Hi,
 the reason for this feature:
i have my main model working with Tbs crossfire tx module in the taranis.
And i have different other quads on SBUS using the internal XJT.

Since we should not power the TX without antenna, i need each time to screw the tbs antenna, power on, change to an internal module.
Is it possible to choose a model as default boot model(1 model working with internal XJT), so that i can power the taranis without having to screw the antenna for the external module each time when powering.
Thanks

geofrancis, 2018-03-04:
Perhaps this should be called "confirm antenna connected before powering external module"

geofrancis, 2018-03-04:
Perhaps this should be called "confirm antenna connected before powering external module"

Graupner radios do something similar where they won't transmit untill you hit ok on the screen.

closed 2019-11-15 by stale[bot]