opentx/opentx #5457: Taranis Q X7S bluetooth trainer data has inconsistent size

garyjohnson, 2017-12-02:
Current version:
Taranis Q X7S
FW: opentx-x7s
VERS: 2.2.0 (63baa76c)
DATE: 2017-09-05 20:07:19
EEPR: 218

Trying to parse the BLE trainer data from the Taranis Q X7S in Slave/BLE mode and running into something odd. When sticks are at min-to-midpoint and switches are off, here's the data I get (from service FFF0, characteristic FFF6):
7e80 **dc5 dc5 e66  dc5 dc3 dc3  dc3 dc5** 8c7e
The length of the data is **16 bytes**.
     
When I move, for example, throttle above midpoint toward it's max
7e80 **dc5 dc5 177d5 dc5 dc3 dc3 dc3 dc5** 6d7e
The length of the data becomes **17 bytes**. It's clear there's an extra byte on the channel value I'm changing (in this case, 177d5 is the throttle value, as was e66 in previous example). If I move multiple channels to max value, I can get additional bytes inserted as well.

Now, my understanding of the data frame (based on bluetooth.cpp in HEAD@2.2) ... https://github.com/opentx/opentx/blob/6f3577c9a580503537b749e263952378f573af8c/radio/src/bluetooth.cpp#L211 ...is that each channel value is packed into a three-byte-wide segment. If so, I'm confused how the channel data suddenly gets an extra byte. Also, without any sort of boundary markers or width counts on the channel data, I'm not sure how you would parse this data correctly. 

I'm guessing this is a bug, but I wanted to see if there's just something simple I'm missing, or maybe if bluetooth.cpp in 2.2 doesn't match the code currently running on the X7S.

schwabe, 2017-12-02:
```
7e80 dc5 dc5 e66 dc5 dc3 dc3 dc3 dc5 8c7e
```

That is already decoded into 12 bit values? Because otherwise that are 20byte

ALso the format is 7e 80 is header, then 8 channel (=12 byte) and then the last 7e is the end byte.

Keep in mind that 7e and 7d are escaped as 7d 5e and 7d 5d.

This seem to be happening  in your second line (17 *7d5d*)

closed 2017-12-02 by schwabe

garyjohnson, 2017-12-02:
I hadn't considered that the boundary markers would need to be escaped if they showed up in the data stream. Makes perfect sense and that is what I was running into. Thank you so much!