opentx/opentx #5460: Horus and X12 (n452)  Companion display no data in Inputs & Mixer screens

coaster007, 2017-12-03:
Hi
Companion 2.2.1 (452), no data displayed in the Input and Mixer Screens, the other screens are o.k.
No problems with Companion 2.2.0

[image]
[image]

3djc, 2017-12-03:
Could you please supply the otx file ?

coaster007, 2017-12-03:
[x10.otx.zip](https://github.com/opentx/opentx/files/1524579/x10.otx.zip)

3djc, 2017-12-03:
Just tested on Debian and official N452:

[image]

Your compilation chain must have issues (please note that we are buildong with Qt  5.7.1)

coaster007, 2017-12-03:
o.k., but i have not compiled the app, just downloaded and installed the deb package form Linux nightly builds.

3djc, 2017-12-03:
I did the same for this test. There seems to be something on your end creating the issue, I would update your QT install to 5.7.1

coaster007, 2017-12-03:
Hi,
i don't now how to update the runtime qt version. after some searches with google i came to the conclusion it's to complicated for me (i am a simple user ;-) ).
Nevermind i can use my mac there is no problem with N452.

tnx and greetings
Graziano

3djc, 2017-12-03:
OK, closing then :)

closed 2017-12-03 by 3djc