opentx/opentx #5467: Timer display in telemetry screen on a X9D+

kennlicht, 2017-12-05:
Using the new RC1 of OpenTX 2.2.1, I've defined two timers an they are displayed on the main screen as they should:
[image]

But please have a look at the telemetry screen:
[image]

I'm not able to see the minutes of both timers, The text is too long. Even if I shorten the text, the numbers aren't shown correctly. That is the case in the simulator an on my radio.

mpaperno, 2017-12-18:
There's no way to fit two custom-named timers in that line and still keep model name and volts the same size.  It's frustratingly close, but just doesn't look good... I tried.  Some options:
* Show up to 3 timers with generic names (`T#` or `Tmr#`).
* If the first one has custom name, then only show that one.
* Reduce font size on model name (would be same size as the timer names). Would still be a bit tight, but should work.
* ....?

kennlicht, 2017-12-19:
I would prefer option 1.

The full name of all timers is already displayed on the main screen. The telemetry screen should display Tht, Ths or Th% or T1, T2, T3, ... Or whatever fits.

PR #5588 (2018-01-03): X9 : Do not display timer name on telemetry page top bar
[image]

Fixes #5467
closed 2018-01-03 by bsongis