opentx/opentx #5468: Default contrast value for X7 in Companion makes it unusable

capitaineflam25, 2017-12-05:
Companion 2.2.1N453 / Taranis X7

Create a new model, open Radio settings : contrast is set to "38" which is more than maximum contrast value on the X7 (30)

Transfer it to the radio, the screen is blank after the splashscreen making the radio unusable.

Adding a limit for this value on Companion could be interesting too.

[image]

raphaelcoeffic, 2017-12-07:
It seems there are two issues here:
- min/max value in "Radio settings" dialog do not correspond to the board's capabilities (X7 has a different range from X9D).
- conversion does not take these values into account either.

raphaelcoeffic, 2017-12-07:
@capitaineflam25 do you have the possibility to compile companion your self to try the fix yourself?

capitaineflam25, 2017-12-07:
Thanks a lot @raphaelcoeffic 
i compile the firmware but not yet Companion, i'll try to do so and will tell you

raphaelcoeffic, 2017-12-07:
@capitaineflam25 in case it gets merged, which would happen faster if you report that it fixes your issue ;-)

capitaineflam25, 2017-12-07:
Companion compiled : i confirm i can't go over 30 for the contrast. If i load an older radio and setting file with an over limit value, it is now displaying "30" on companion.

I guess this is what you expected ? so thanks a lot.

[image]

raphaelcoeffic, 2017-12-07:
@capitaineflam25 yes, that’s exactly what it should be. Could you please try to transfer the converted file to the X7 without first opening the radio settings dialog? That should work as well.

capitaineflam25, 2017-12-07:
Unfortunately that is not OK ; 

i load an old converted file on companion with a value of 35 for the contrast

then i flash the radio, and the screen gets blank

if i read back the data from the radio i still can see "30" for the contrast on Companion, but i guess it was truncated after reading it from the radio...

raphaelcoeffic, 2017-12-07:
Exactly, as soon as the settings dialog opens, the value is clipped.

raphaelcoeffic, 2017-12-07:
@capitaineflam25 strange enough.... could you please upload your Sky9X model file? (before the conversion) I could not reproduce it with my X9D profile with contrast set to 45. My X7 would then start with the contrast set to 30.

capitaineflam25, 2017-12-07:
The file was one saved previously saved from a Companion without your fix.

If i read my sky9x from your latest patched version and write it directly to my X7 i get a darker screen with a contrast set to 30 👍 

So my previous test was not good, the issue is fixed with your new code.

PR #5473 (2017-12-07): added support for min/max contrast to eeprom interface and use it
fixes #5468
commit 93457811 (2017-12-07): added support for min/max contrast to eeprom interface and use it (#5473)

Added support for min/max contrast to eeprom interface and use it. Fixes #5468
closed 2017-12-07 by bsongis