opentx/opentx #5476: Add the ability to see Logical Switches (LS) 33-64 from the main view screen. XD9+

Kennamer, 2017-12-07:
Taranis X9D+

On the 3rd page of the main view screen (the Logical Switches screen), only switches 1-32 are visible (LS 1-32).

It seems intuitive that by pressing the + key, you would then see 'page 2' of the logical switches: LS 33-64, just like how the Channel Monitor screen operates.

Can this be made to happen?

Thank you.

Kennamer, 2017-12-09:
Hello mpaperno,

This was my first experience posting on here and I have to say I am impressed! Addressed and fixed in less than 24 hours! Thank you.

I just have a small additional request: Can you make the first page read LS 01-32 (adding the zero to the one). It will make the characters stay in the same place on the screen when switching pages and also corresponds with the actual numbering of the Logical Switches (01, 02, etc.). It is just for aesthetics, but thought I'd ask since all you (probably) have to do is add a zero. ;-)

Thanks again for the quick fix!

Kennamer, 2017-12-09:
Looks great! Really blown away at how fast it was all handled.

Thank you! 

I suppose I close this thread now, so here it goes...

3djc, 2017-12-09:
Thread will be automaticaly closed if/when the correction gets into OpenTx

PR #5486 (2017-12-09): [X9] Allow LSs 33-64 in main view & fix X9E view of channels 17-32
+/- keys or left/right encoder will switch between LS groups 1-32 and 33-64.
Also with X9E (encoder only) there was no way to navigate in channels/mixes monitor to the higher channels.

Closes #5476

[image]
[image]
commit dec630ef (2017-12-09): [X9] Allow LSs 33-64 in main view & fix X9E view of channels 17-32 (#5486)

* [212x64] Main view toggle LS groups 1-32 and 33-64 with +/-/encoder. Closes #5476

* [212x64] Also allow encoder events in channels view for switching channels range (fix for X9E).

* Pad LS range numerals to 2 digits.
closed 2017-12-09 by bsongis