opentx/opentx #5488: Volume setting not transferred properly between Companion and TX (X10)

x10user, 2017-12-09:
On my X10 I assigned the RS for volume control. After reading the TX settings to Companion the parameter for volume control was not transferred.
[image]

I tested it also the other way. If the parameter is set to RS in companion volume control works on the TX (after writing settings to the TX), but in reverse (down make it louder).
Nevertheless, the Parameter-field for volume on the TX is empty. After re-assign RS the volume control works as expected (RS up for volume up).

OTX-file for reference:
[FrSky X10.zip](https://github.com/opentx/opentx/files/1544627/FrSky.X10.zip)

x10user, 2017-12-09:
The issue was observed for a while, latest build tested: N457 (both on Companion and TX)

mpaperno, 2017-12-09:
Turns out the X10 LS and RS analogs are being handled differently in firmware vs. Companion, so this is the underlying issue.  Companion thinks X10 has L1/L2 sliders.  Under further investigation now.

PR #5493 (2017-12-10): [Companion] Adjust X10/S hardware options to match radio (no L1/L2/JSx/JSy)
Removes L1/L2/JSx/JSy as possible inputs on X10, including in radio HW and calibration settings.

Fixes #5488  (tested only in sim)

Does not deal with any conversion issues.  Especially the "silent" conversion between X10 and X12, which is not currently detected at all.

When going from X12 to X10, L1/L2 become LS/RS, and vice-versa when going the other way (same issue with any X9E/X12 <-> X9D/X10 conversion).
commit e4380754 (2017-12-10): [Companion] Adjust X10/S hardware options to match radio (no L1/L2/JSx/JSy). Fixes #5488 (#5493)
closed 2017-12-10 by bsongis