opentx/opentx #5489: Add "View notes" on telemetry screens menu

capitaineflam25, 2017-12-09:
I'm both a fan of the ability to display notes and custom telemetry screens.

Would you see an interest in invoking the "View notes" from the telemetry screen menu ? 
So you don't have to exit Telemetry with an [EXIT] long, and get back in it with [PAGE] long

[image]

capitaineflam25, 2017-12-31:
The complete "Reset" menu would be usefull here too.

When you need to reset your timer to prepare a new flight, you need to exit telemetry screen, do the reset and go back to the telemetry screen.

kilrah, 2017-12-31:
> When you need to reset your timer to prepare a new flight, you need to exit telemetry screen, do the reset and go back to the telemetry screen.

The "reset flight" entry is supposed to be used there...

capitaineflam25, 2017-12-31:
The reset flight is more complete.
I tried a « Reset flight » after a battery change and it also launch the « switch warning ».

I was wondering if the classic home screen and the telemetry screen should have the same menus since more pilots like to have custom LUA telemetry screen.

This could take more sense if #5156 is developed.

closed 2019-11-15 by stale[bot]