opentx/opentx #5491: Can't get trainer system to work on X10 w 2.2.1RC1

ducatiJake, 2017-12-09:
I am trying to get a Skyzone FPV goggle head tracking input into my X10 and cannot get it to respond at all.  I have an X7 on 2.2 which seems to work fine.  First of all, when I plug in the Skyzone into the X7 I get a voice telling me 'Trainer signal recovered', remove cable and get 'Trainer signal lost'.  I don't get that on the X10.  I have gone into a new model and set Trainer Port Mode: Master/Jack.  Inputs 5 and 6 set to TR5 and TR6.  Mixers CH 5 and 6 set to I05 and I06.  Added a special function on SA down - Trainer --- checkbox checked. On my X7 I see channels 6 and 7 move around as I move the Skyzone headset.  Nothing on the X10.  Is there some other trick to enable trainer on X10 I am missing?  I can see the movement in the monitor main screens of the X7.  Should I see them in the monitor screens of the X10?

raphaelcoeffic, 2017-12-09:
yes, should work the same.

pettert85, 2017-12-11:
I can confirm this.  Connecting my head tracker to my x9d+ works great, but on my X10S running opentx 2.2.1RC1 there is no movement in channel monitor when connected to the jack next to mini usb.

schwabe, 2017-12-11:
Has any of you tried if the trainer works in FrOS?

pettert85, 2017-12-11:
No sorry, I love openTX so much i installed it two hours after i received my new radio. I would think a lot more people complained about it if it did not work in FrOS though. headtracking is quite popular .

schwabe, 2017-12-11:
Would be good to know if this a sw or hw issue

kilrah, 2017-12-11:
Trainer verified to work fine with X9D as source. You might want to try that. If it does, then it's probably a hardware incompatibility with your headtracker.

ducatiJake, 2017-12-11:
While trying the head tracker with the X7 I have which worked, but not on the X10 I also tried using the X7 as a slave to the X!0.  This also did not work.  Not sure what would be different between X7 and x9D, but by coincidence I also have an X9D, I can try that tomorrow.

ducatiJake, 2017-12-12:
OK, learned a lot about the weakness of trainer connections today.  This is definitely NOT a SW problem in OpenTX.  I tried using X10 as master and X9D as slave.  Got very random connectivity, but when connected, it worked.  Had to have cable partway disconnected from X9D to get connection.  OK, that sucks.  kilrah who verified X9D performance, is your cable a mono or stereo cable?  Mine is stereo and I get random connection.  The reason I used a stereo cable is that the connection on the head tracker I have is a stereo cable, I figured that was the standard.  I looked inside the X10 and X7 and they both have stereo connectors, tested all 3 and they were all consistently good, but when plugged in completely they did not act as a trainer.  I have found in RC Groups many posts which say both mono and stereo cables work and some that say they don't.  I suspect that the X10 and X7 and probably the X9D (although that connector is hidden and I can't see it) while they use stereo connections only use 2 of the wires for signal and gnd and assume you will use a mono cable which will hit 2 connections with gnd and only one with signal.  Just guessing here cuz I don't have a mono cable to test this theory.  By the way, I got the same effect with my head tracker.  If I partway put the cable into the X10, I can get connected and the head tracker works.  I will find a mono cable and see if I get a consistent connection.  Bottom line, not OpenTX's problem.

schwabe, 2017-12-12:
You can also make a mono cable from a stereo cable by cutting the cable and  soldering the GND to the right channel together. And yes the X9D+ has only one of them connected.

pettert85, 2017-12-12:
Does this mean i have to modify my cable.?  I'm using the trinity head tracker from fatshark.  Is the jack on x9d+ and X10 different? I'm confused.

ducatiJake, 2017-12-12:
I ordered a stereo to mono adapter from Amazon.  I'll let you know if it works for my headtracker.

closed 2017-12-12 by kilrah

pettert85, 2017-12-17:
Just tried with a mono/stereo adapter and now it works. Thanks.

pettert85, 2017-12-17:
It is weird of Frsky to change a already working stereo jack to a mono jack btw.

raphaelcoeffic, 2017-12-17:
@pettert85 I think it has always been meant to be a mono cable, just the internal connector was slightly different and would somehow support a stereo cable, which was neither intended nor tested. But that's just a wild guess.

ducatiJake, 2017-12-17:
[image]

Differences between mono and stereo conectors

ducatiJake, 2017-12-17:
I learned way more about mono/stereo connector variability than I ever wanted to know in the last several days.  I bought a stereo -> Mono adapter (on left) which I tried with the stereo connector coming from my headset with little improvement in connectivity from just using the stereo jack.  I also bought an actual trainer cord, which is just a mono male - male extender, that is on the right.  The center is of course the original stereo jack originally  on the headset.  Look how similar the trainer and stereo jacks are, and how dissimilar the mono converter jack is.  Much wider insulator area.  I also bought a mono jack that I could solder onto my headset cable to replace the stereo one, which I tried and got inconsistent connectivity.  That jack is not in the picture, but looks identical to the mono jack on the left which also does not work.  I am getting around this in my case by connecting both the bottom and center conductors in the stereo jack to the gnd connection from the headset and just the tip to the signal from the headset.  I my case this seems to be reliable..  We are getting burned by these manufacturers not following standards.  FrSky should not be using a stereo connector in their radios.  I am sure this was long ago spec'd as a mono connector.  They are little used these days and I bet FrSky can get stereo connectors for 2 cents less than specifying mono.  The headset guys are doing the same thing, not following a standard in place for many years,  And the manufacturers of the jacks seem all over the place as to what the mechanics of a mono jack are supposed to be.  Your mileage will vary with these connections.  Not to bitch too hard at FrSky, I bet most of the radio manufacturers have this randomly implemented.

kilrah, 2017-12-17:
When I tested to validate I used a 4-connector cable and it was fine... but then again it might even vary between batches, who knows.

Anyway jack is a mess like many other many-decade-old connections. Still is the most common trainer connection in anything RC from the past 30 years or so, so it's kinda normal it's still used for legacy compatibility...

ducatiJake, 2017-12-17:
No argument there, each vendor had it's own style of trainer cable, there isn't really quite a standard in RC.  What surprised me most was the different styles made of mono audio connectors.  I'm sure that's a real documented standard, probably RCA or the like.  Then again, I'm sure the ones I got were cheap Chinese knock-offs and all bets are off.  The real lesson is check your stuff, wiggle the cables, if it's seems loose or drops out, make it right.  I've been flying for many years and this is the first time I have used a trainer cable setup for anything.  I have seen many training accidents and confusion about trainer cords working and not working and sometimes working over the years.  Maybe I can add some knowledge to some of the confusion at my field at least.

Gaz-B, 2020-01-17:
I know this thread is long dead but seems to be missing a conclusion. i have just needed to get it working and found ground to middle ring and PPM to tip with no connection to base on a stereo jack on the X10S.

ducatiJake, 2020-01-17:
Hi Gaz, This is pretty random stuff these days.  I think best bet is to use a mono cable, but manufacturers seem to vary on the design of this connector.  Sadly you have to try different cables till you find one which works for your situation.
Good Luck, John

Gaz-B, 2020-01-21:
Performing perfectly on 2.3.5