opentx/opentx #5492: Scroll wheel issue

PinkJackal, 2017-12-09:
I have a problem with the scroll wheel on my Horus, it's working erratically in Open-TX, if I scroll down, it will scroll down than jump up, than down again, the same if I scroll UP, it will go UP than down, than UP again.

With FrOS this is not happening, I have reinstalled Open-TX 2.2.0 3 times now, this is still happening, very annoying!

raphaelcoeffic, 2017-12-09:
@PinkJackal Do you mean that it jumps at the beginning of the menu when you reached the end of the list? If this is the case, this is absolutely normal.

PinkJackal, 2017-12-09:
NO! I mean, when I scroll down for example, it doesn't go down consistently, it will jump up 2 or 3 rows, than will come down again. I hope you understand, it's hard for me to find my exact words in english, if you want I will make a video!

raphaelcoeffic, 2017-12-09:
Seems quite odd indeed... Is that happening regardless of the speed at which you turn the wheel?

PinkJackal, 2017-12-09:
yes. At first I thought I have a bad scroll wheel than I checked with FrOS and it's never happening, not even once

x10user, 2017-12-09:
I can confirm that the odd scroll wheel operation (values jump back and forth sometimes) is also present on my X10. I was afraid that it might be caused by a hardware issue (encoder) but it might also be the same behavior as described above. I didn't cross-checked with FrOS.
If needed I'm willing to perform further testing on my X10.

raphaelcoeffic, 2017-12-09:
well, I cannot reproduce it on my X10. So yes, I'd probably need additional data. Just not sure what at the moment. Maybe a video or something like that, so I can see how it manifests?

PinkJackal, 2017-12-10:
FrOS- as you can see from the video, everything works as is should
https://www.youtube.com/watch?v=Bk9F-wZuSl4&feature=youtu.be

OpenTX- you can see how the line jumps as I scroll in one direction, this also happens thru the menu.
https://www.youtube.com/watch?v=gmdSnSHBMH4&feature=youtu.be

schwabe, 2017-12-10:
The links tell me that the video is not available, wrong sharing settings?

PinkJackal, 2017-12-10:
https://youtu.be/Bk9F-wZuSl4
https://youtu.be/gmdSnSHBMH4

raphaelcoeffic, 2017-12-10:
Now I can reproduce it on my X10! The Volume slider seems to be a good test. Did you notice the issue on other things than sliders?

PinkJackal, 2017-12-10:
I pick the sliders so you can see it better, but for me it is happening on menu scroll up or down and on numbers as well

raphaelcoeffic, 2017-12-12:
Seems a bit strange indeed. I tried all the usual tricks (in the key driver), but the effect is still there... The sampling rate is 1KHz, as usual for these kind of encoders.

raphaelcoeffic, 2017-12-13:
I've spotted the exact same issue on the X7 as well, whereby it's really hard to spot, as the screen refreshes way slower than the TFTs used in the X10/X12S. The same fix can probably be applied to the taranis key driver, as it is 100% identical to the horus one.

PinkJackal, 2017-12-16:
Do I have to download this fix or it should be available in next OpenTX release?

raphaelcoeffic, 2017-12-16:
@PinkJackal the fix is scheduled for 2.2.2.

Devil-67, 2017-12-16:
@raphaelcoeffic  
maybe it's possible If you can pack the fix in 2.2.1 in a nightly, would like to test it already ;-)

raphaelcoeffic, 2017-12-16:
@Devil-67 it’s not mine to decide ;-) But I can upload a compiled firmware. What options do you guys use? LUA, GVARS, NOHELI, and what else? MULTI maybe?

raphaelcoeffic, 2017-12-16:
@Devil-67 this should be sufficient for you to test: https://github.com/opentx/opentx/files/1564645/X12S_rot_enc_fix.zip

raphaelcoeffic, 2017-12-16:
For the others with a X10: https://github.com/opentx/opentx/files/1564654/X10_rot_enc_fix.zip

Devil-67, 2017-12-16:
You are the best !!!!!!!
The problem is solved with the jumping in the menu is fixed !!!
Finally, you can make adjustments in the Horus X12 without unintentionally adjusting something ..
Juhuuuu I'm happy ;-))

I was already there and wanted to replace the encoder, because I thought he is broken.
It's time again for a small donation!
If now # 5172 is being considered Christmas is saved.

keep up the good work

Merry Christmas guys

Devil-67, 2017-12-16:
Donated

raphaelcoeffic, 2017-12-16:
@Devil-67 good to know it works :-) Concerning your other issue, I'm afraid it requires an EEPROM format change, which cannot be done until 2.3...

Devil-67, 2017-12-16:
No problem, I'm happy if it does not fall under the table because there are always more sensors Especially with OXS (OpenX Sensor)
Then the RB's also send a lot of data even with sub ID's.
If it is feasible in 2.3 good, and if it is possible earlier, the top class ;-)

Thanks again

PinkJackal, 2017-12-16:
Perfect, thank you for your hard work!

closed 2017-12-18 by bsongis

tonythetiger95, 2018-12-18:
I have a Taranis qx7 that has the same erratic scrolling. Will opentx firmware update fix it?

Excaliburst, 2024-03-25:
Hi guys, I am on Q X7 and confirm this erratic rotary encoder issue. I have upgraded to FW 2.3.15, and still see this issue.
Please help. It shouldn't be an encoder issue, when so many people see the issue... Or...

Any help appreciated.

PR #5510 (2017-12-18): improved Horus rotary encoder driver and generic behavior
Heavily tested on X10 and X7.

fixes #5492