opentx/opentx #5498: Reboot during flight X9D Plus 2.1.7

dallinbriggs, 2017-12-11:
I have a Taranis X9D plus radio running 2.1.7 that hasn't had any issues for years and suddenly today my radio rebooted during the launch of a discus launch glider. The receiver went into failsafe for a few seconds and after landing I noticed an exclamation mark in a black box on the main screen after landing. Also, the audio wasn't working after that. I'm assuming that is booted into a 'safe' mode to boot faster. I haven't been able to recreate the issue after that yet, but the audio works after cycling power on the radio.

I didn't see any documentation on the issue yet. Is this a new issue or is there existing documentation that exists that I wasn't able to find?

powercycle, 2017-12-11:
Have you updated the firmware recently?  Although I'm not a developer, seems to me that if you haven't updated anything, then this is likely a hardware issue. I'm using a X9D+ and am not experiencing those sort of issues.

kilrah, 2017-12-11:
It is indeed the safe mode that's intended precisely for these cases.
Since it happened during launch you might want to check your battery wiring and connections. Could also be caused by static electricity discharge.

dallinbriggs, 2017-12-11:
I haven't updated the firmware recently. Static discharge seems unlikely since it was in my hand the whole time. Would a momentary glitch in power cause it to reboot like that? I thought that would cause the radio to do a full reboot. Not necessarily a safe boot.

schwabe, 2017-12-11:
yes, we set a flag, in eeprom if power off was from us. If we don't read that flag on startup, we go into failsafe mode. You can easily test this yourself by removing the battery while the TX is on.

powercycle, 2017-12-11:
OK, that is cool.  Didn't know that existed.  So if the battery disconnects, and reconnects, it will reconnect signal but be in "safe mode"?  What happens in safe mode?

3djc, 2017-12-11:
You skip all warnings, no lua, no sd, but you have control of your model

dallinbriggs, 2017-12-11:
Well I know it doesn't load any of the sounds from the SD card so it can boot faster. I'm not sure what the other features it doesn't load in safe mode, but none of them seem flight critical. Thanks everyone for your input. This gives me enough reason to believe that is may be a hardware issue. I'll continue to investigate what the problem is and see if I can duplicate the event, but for now it seems like it might not be related to the firmware. I'll close the issue and I'll only reopen it if I can validate that it is related to the firmware. You guys are doing great!

closed 2017-12-11 by dallinbriggs

powercycle, 2017-12-11:
So it includes internal and external module.  Makes sense that it would, but didn't want to assume.