opentx/opentx #5502: Disable antenna warning for Taranis X9D+ A01

LapinFou, 2017-12-11:
Hi folks,
Since the **bindopt** is now part of OTX2.2.1, could you add a check using XJTVER in order to disable antenna warning for the Taranis X9D+ A01.
This has already been discussed here: #5038 

I opened a topic on the French forum to collect some information http://frskytaranis.forumactif.org/t6393-a-tous-j-ai-besoin-de-votre-aide-help
I didn't have much answers yet, but it seems the XJTVER for all radios with a functional RAS is always 512.

LapinFou, 2017-12-11:
I opened a post on RCGroups in order to collect more data:
https://www.rcgroups.com/forums/showthread.php?3002032-OpenTX-2-2-1RCx

LapinFou, 2017-12-13:
I realized my request was not complete.
In fact, could you add a check to disable the antenna check when the XJTVER is **1025**.
I tried to dig in the code, from my poor understanding, the file to modifiy is this one:
_radio/src/telemetry/frsky.h_

Since a check is already existing for X9D+ & X9E, could you kindly add a 3rd condition for XJTVER **1025**??
```
#if defined(PCBX10)
  #define IS_SWR_VALUE_VALID()         (false)
#elif defined(PCBX9DP) || defined(PCBX9E)
  #define IS_SWR_VALUE_VALID()         (telemetryData.xjtVersion != 0 && telemetryData.xjtVersion != 0xff)
#else
  #define IS_SWR_VALUE_VALID()         (true)
#endif
```
That would be awesome. Thus I will be able to update my XJT to the latest version and get the benefit of the new **bindopts** feature.

kilrah, 2017-12-13:
My A01 X9D+ has SJTVER 1025 and SWR works, so no...

LapinFou, 2017-12-14:
What?? I'm very surprise (and disappointed), since I thought all X9D+ A01 were supposed to have a none-working SWR.
Silly question but are sure that your A01 Taranis is really a X9D+, not a X9D?

Anyway, this is a very bad new. 😭 

Plan B, would be to add a compilation option such as **noswr** in order to disable this check. What do you think?
Otherwise all owners of a X9D+ without a working SWR will never be able to used the **binopt** features. Or, they must buy an new expensive RF-backboard. 😞 

By the way, I got a (not helpful) answer from FrSky:

> Hi BR,
> 
> Thanks contact us, the XJT hardware of A01 don't support SWR function, so now you can replace the new backboard for this function, or you can request to the OpenTX team to disable the SWR/RAS check on A01 VERSION of xjt, then flash the new firmware for your Tranis X9D+ version A01.
> 
> Best regards
> Ray.

kilrah, 2017-12-14:
What's the reason for not being able to use the bindopts? Mine runs it fine...
Yes it's a Plus, amber LED version. 

Given the mess FrSky have made with their batches and versions manually disabling might indeed be the solution in the end.

LapinFou, 2017-12-14:
To use the **bindopts**, I need to flash the internal XJT board with the latest firmware (otherwise it simply does not work).
Doing so, the **binopts** features will work. But, that it the whole point of my request, with the latest firmware I get "defective antenna" errors.
With the original firmware, my internal XJT will respond 0 to the _XJTVER_ request.
With any others firmware versions, _XJTVER_ will return 1025. Thus, OpenTX will think "Hey this a XJT board with a valid SWR! Let's check the SWR value...". 

Here are the results coming from my both A01 X9D+:

Version|RAS|XJTVER|Comment
-|-|-|-
``XJT_130725`` |97 |1025 | **Antenna Defective** error
``XJT_141016`` |0 |0| No **Antenna Defective** error
``XJT_161214`` |96 |1025 | **Antenna Defective** error
``XJT_170317`` |98 |1025 | **Antenna Defective** error

I want to use the **bindopts** features (which will be part of OTX 2.2.1 as you already know).
So, all A01 X9D+ owners (version with the blue backlight) as myself got only 2 solutions:
1. A **noswr** option in order to use latest XJT firmware.
2. Buying a new RF backboard.

kilrah, 2017-12-14:
We said previously that it was FrSky's problem to continue reporting the proper XJT version in newer firmwares... or at least tell us which new ones should be included to the exclusion list if any. 

Not feeling like doing investigaiton work to try and figure it out...

I get a RAS of about 30-35 with my module on latest FW. I seem to remember that there were 2 different batches of A01 which only the first having the issue...

MikeBland, 2017-12-14:
My prototype plus also reports around 33 with a XJT_VER of 1025.
Note that any "noswr" option would need to be able to distinguish between and internal module and an external module, and you could be using an old external module that doesn't support SWR and a new one that does support SWR.

LapinFou, 2017-12-14:
@kilrah 
RAS of 30-35 seems to me like a none functional SWR. Thus, it could make sense to add the XJT version 1025 in the black list.

@MikeBland 
Good point indeed. 👍

3djc, 2017-12-14:
Why non functionnal ? 30 is a perfectly valid value, below the alarm threshold of 51

LapinFou, 2017-12-15:
I thought that a normal SWR (without shielding the antenna) should be below 10.

@kilrah & @3djc 
Are you OK to implement a **noswr** compilation option for the internal XJT?
I will be so grateful to you if you agree.
tight now, I'm very disappointing to not be able to use the _bindopts_ which will be part of the final 2.2.1 version.
Please???
[image]

kilrah, 2017-12-15:
> I thought that a normal SWR (without shielding the antenna) should be below 10.

That's why we're now hiding it away, people misinterpret it all the time.
As long as you don't have the warning it's considered OK.

LapinFou, 2017-12-15:
Yep, I understand.

LapinFou, 2017-12-17:
Hi folks,

Any chance to accept and implement this enhancement request for OpenTX 2.2.1 final?
It would be so nice. 😄 

Otherwise X9D+ A01 (blue backlight version) owners won't have all the benefits of this new version.

3djc, 2017-12-17:
Since we just released 2.2.1 final, I'm afraid  that's a no

Maybe you can use 2.2.1 with a xjt firmware returning valid info. yes, your choice of binding options will be inecfective, and has to be done manually using jumper if different from default, but it is really for FrSky to fix their xjt firmware

LapinFou, 2017-12-17:
I understand, but I send several email to FrSky and the only relevant answer I got is: buy a new RF backboard. 😢 
Moreover, the only way to disable the telemetry on a XSR RX IS with this **bindopt** option...
So, having a **noswr** compilation option is really my only hope to get something done quickly.

Maybe it could be part of the 2.2.2 version??

bsongis, 2017-12-28:
I take it, no need to buy another board just because of a false warning

LapinFou, 2017-12-29:
@bsongis 
Thanks you!! Thanks you!!!
You made my day. 😄 
Did I thanked you? 😉 
Thanks again. This is a very good news to start a new day.

schwabe, 2017-12-29:
@bsongis shouldn't we name it noras for consistency?

kilrah, 2017-12-29:
Agreed

bsongis, 2017-12-30:
PR changed, `noras` instead of `noswr`

closed 2018-01-02 by bsongis

LapinFou, 2018-01-04:
I tried OTX 2.2.2N479 with the **noras** option.
This is working fine. 👍 
When I went back to OTX 2.2.1, I got no more _Antenna Defective_ error! 😱 
So, somehow, something change about the RAS management between OTX 2.2.0 and OTX2.2.1.
I tried to upgrade the XJT FW on my 2nd X9D+ A01 (OTX 2.2.1), same result. No more _Antenna Defective_ error.
I thought it was worth to mentioned it.

3djc, 2018-01-04:
So basically you are saying 2.2.1 works and this in unneeded ?

kilrah, 2018-01-04:
Did you actually check the number?
It could simply be moving/unstable and today is lower than the threshold...

LapinFou, 2018-01-04:
I can't. The RAS value which was displayed in ANALOG TEST menu is not visible on OTX2.2.1.

LapinFou, 2018-01-04:
[EDIT] I just tested again, and the RAS is displayed as it should. The value is 31/32...
I don't understand what is going on....

LapinFou, 2018-01-04:
I tested on my 2nd X9D+, the value is 49. If I'm playing a bit with the antenna, then I got the alert message.
So, it seems the hydrometric has got a non negligible impact on non functional RAS.
So, forgot what I said before and sorry...

LapinFou, 2018-01-04:
OK, I found my mistake. When I have done my tests, the active model was one using the Multi-4in1 in the external bay. It is why the RAS/XJTVER was not displayed.
So, I properly tested again with OTX 2.2.2N479 and the **noras** option.
This time, the XJTVER is displayed (1025 in my case) and the RAS is coerced to 0 as expected.
So everything is perfect.
Again, thanks to @bsongis !
Can't wait OTX 2.2.2 😄

throttlenerd, 2019-10-11:
> My A01 X9D+ has XJTVER 1025 and SWR works, so no...

Yes! My X9D+ (Pre 2019) showed correct SWR too! But after updating to 2.3.1 I can't find it, will dig further..

commit 269831e0 (2018-01-02): Fixes #5502 (#5575)

* Fixes #5502

* RAS renamed to SWR