opentx/opentx #5505: SD Memory Corrupts

jcutcher01, 2017-12-12:
I am new to OpenTX and Horus X10. Today I installed Companion 2.2.1 RC2 and it flashed 2.2.1 RC1 firmware to my Horus X10. I spent about 2 hours setting up one model along with all sorts of personalized settings. I then opened companion, with the radio on and connected to USB, clicked Read Models and Settings from Radio and everything appeared on the screen. I then added some new settings and wrote that data back to the radio and then I ejected the SD card. I powered off and then back on. A memory corrupted error message appeared and all of my data was lost. The radio started over from scratch asking me to calibrate the sticks, etc.

I then tried several more times connecting to companion and reading and writing and every time the SD card gets corrupted.

Hopefully I am just doing something wrong.

Jeff

chiron7000, 2017-12-12:
I already reported this yesterday evening, see #5504.

2.2.1 RC2 has also been removed from the download page: http://www.open-tx.org/downloads

kilrah, 2017-12-12:
Do not use RC2, there's a reason it was removed (and it's exactly that).

closed 2017-12-12 by kilrah