opentx/opentx #5516: Companion 2.2. Assign download behaviour to profile

mpjf01, 2017-12-14:
The new arrangement for election of type of firmware to download, where the user can choose between the three options "Use releases (stable)" and so on - applies the choice to all the user's profiles. I have several radios and would like to be able to have a persistent selection applicable to the profile attached to each. So, for example, for my main flying radio I would prefer to choose the stable builds, for a test radio I might wish to download the nightly builds. 

Is this a reasonable change to request or would it not be supported by the current data structure?

3djc, 2017-12-14:
We made the change to have one setting for both companion and firmware : we want to ensure you are using a version of companion that is designed to handle your radio data. You propbably have noticed that Companion has gone through a massive revamp and does now much more than in the past, but that means it needs a tighter link than before.

In your scenatio, I think the best would be to have 2 companion install

kilrah, 2017-12-14:
Except that won't work since both would be using the same settings.

3djc, 2017-12-14:
Hum true, we need to think about instanciating those then !

mpjf01, 2017-12-14:
I'm obviously not familiar with the intricacies of the system so would appreciate an explanation of how having only one choice of firmware download options for all radio profiles materially assists with the objective of ensuring that the Companion version matches the radio data. I'm not challenging the statement, just curious about the relationship.

I thought that the concept of profiles was, in part, to assist with maintaining multiple transmitters, so keeping that element of the functionality would be better if possible. Having said that, if the only option is to maintain two or three instances of Companion for the most recent firmware version, plus another for any previous versions (if desired), is doable, it just adds to the workload for the user whenever they have to be updated. 

Right now the profiles are common to all versions (I think) so that would need to be changed or some method developed to help the user avoid selecting the incorrect version of Companion for the radio being used at the time.

The average user probably doesn't have multiple transmitters or require different versions of the firmware on each.

3djc, 2017-12-14:
It is not about ensuring all transmister are the same, it is to ensure transmiter and companion are on the same page. Lets run a couple of scenarios of why we merged the choice for companion and transmiter first:

- you want to run release/rc companion and nightlies radio (very strange choice, but why not): new version of data model comes, radio get it, not companion. Result: companion cannot edit the radio anymore, since the data are all garbage to it

- a more likely scenario : you want nightly companion and release/rc firmware. New data model comes, you open your otx / get data from radio, gets converted, you can deal with it in companion, but you have no way to send it back to radio (it would not understand it)

Thats why you need to have your radio fimrware and companion on the same 'branch', and in extend, why you cannot have a single one for companion, and multiple for radios

Hope thius clarifies it

mpjf01, 2017-12-17:
The relationship between the firmware and the transmitter are, for the most part, defined by the profile data. This includes the firmware version specific to the hardware defined in the profile. The exception is in the application settings where the user can specify the firmware at a higher level - release version, nightly builds and so on. I'm sure that there is a reason but I can't help wondering why this higher level selection cannot also be in the profile? The need to manage multiple instances of Companion does not seem to me to be the ideal option.

3djc, 2017-12-17:
Because you thinking multiple instance of rhe same companion, but thats where the issue is, you need different companion for a nightly and release. I'll try to explain again with an exemple. Let's say you are running 2.2.0 on one of your radio, but you want latest nightly for another. So you have to download a nightly companion for that. So far, so good. 

Now you edit your 2.2.0 radio data with a recent nightly companion : you could setup your external module to R9M, companion nightly. Then you send that data back to the radio, but the thing is,a 2.2.0 radio doesn't have any clue how to deal with R9M module settings, and that might lead to unpredictable results.

For this reason, verison of companioin and radio need to be kept in sync as much as possible

mpaperno, 2017-12-17:
Ideally each new Companion version would be backwards-compatible with older firmwares (to a point), but this is not very realistic given all the variables (pun intended).

You can already run minor versions simultaneously, with different settings (eg. 2.0, 2.1, 2.2).  So while a bit awkward, that's a workable solution.

For "patch" versions (eg. 2.2.x) there could be a couple solutions... we make sure they're all backwards compatible with the last minor (maybe starting with next one 😉 ), or we give each of those patch versions their own set of unique settings as well.  (Perhaps improved settings import/export could be added to facilitate the latter option a bit.)

kilrah, 2017-12-17:
Within a given major version companion should already naturally not break anything. If something is added  it is in bits that were spares and an older companion of the same major version would not touch them. 

Normally what you're asking should work technically, just not sure we want to encourage that. Recommended way would be to install release companion when you want to manage a release radio and install nightly/RC when you want to manage a radio on nightly/RC.  Takes 15 seconds if you've got the installers at hand.

mpaperno, 2017-12-17:
But there's e.g. 2.2.0 release and 2.2.1 rc/nightly, both using the same CPN settings. So can't run both of those together... right?  😕 

> Within a given major version companion should already naturally not break anything.

Major?  As in going from 2.0 to 2.2?  I think you mean within minor (2.2.0 to 2.2.1)...?  In that case you're probably right, except in a case such as @3djc mentioned... which is a valid point.

kilrah, 2017-12-17:
> Major? As in going from 2.0 to 2.2? I think you mean within minor (2.2.0 to 2.2.1)...?

Wording... I mean "if the major version is same" yes.

> But there's e.g. 2.2.0 release and 2.2.1 rc/nightly, both using the same CPN settings. So can't run both of those together... right? 

Well settings shouldn't differ since it's the same major version... Only thing is when you run release companion it will reset the download pref to release, so when you then run the nightly you need to set it again...

mpaperno, 2018-07-21:
Companion 2.3 has a more flexible settings structure and adds settings import/export functions. So there's a good opportunity to do something to facilitate having multiple versions installed.

If we're continuing the paradigm of having breaking changes in each `RELEASE` version (as in `MAJOR.MINOR.RELEASE`) then the most obvious choice would be to make the settings location specific to the release version (instead of minor like it is now).

(Personally I'd vote for going back to the `MINOR` versions having bigger "potentially breaking" changes and only use `RELEASE` versions for bug fixes or small changes which do not affect radio data structure.)

> Only thing is when you run release companion it will reset the download pref to release, so when you then run the nightly you need to set it again.

I'm not sure if/when this changed, but in 2.2.2 (at least) that's not the case... it'll actually keep your "nightly" setting even for release/rc version.  But release/rc version will never check for nightly updates.  Which is confusing so in 2.3 I added a warning in the settings UI when that happens.

kilrah, 2018-07-21:
> Personally I'd vote for going back to the MINOR versions having bigger "potentially breaking" changes and only use RELEASE versions for bug fixes or small changes which do not affect radio data structure

That's what we do and always have.

closed 2019-11-15 by stale[bot]