opentx/opentx #5520: TGY 9XR PRO-OpenTx 2.2. – the value of the RAS is not displayed.

JASSoft1, 2017-12-15:
The value of the RAS is not displayed (indicated). To be on the side of the 7/9 "Analogs"??
The value of the SWR was already sorted. For the correct setting of the antenna TX is the value of the RAS (SWR) important.

kilrah, 2017-12-15:
What module?

JASSoft1, 2017-12-16:
XJT

JASSoft1, 2017-12-16:
I'm sorry I accidentally closed the #5520.

AKA00TG, 2017-12-23:
My Qx7 also has no RAS in the analogs  (orSWR in the parameter select) and I do need to see it, if only just once. Is there an ID number or some other way to see this value?
open tx 2.2.1      Taranis Qx7

schwabe, 2017-12-23:
For the QX7 that is by design since the X7 XJT has no SWR/RAS measurement.

closed 2017-12-25 by bsongis

JASSoft1, 2017-12-25:
I'm sorry, but at my 9XR Pro with OpenTx 2.2.1 the problem persists ..
I add the configuration: 9XR Pro + XJT (version 2015) - D8 mode, D6FR receiver, or D8R-II PLUS, or F801 .... Telemetry works normally. On screen 7/9, RAS is not displayed.
In version FW OpenTx 2.1.9 (in the same configuration). the SWR value is displayed normally

3djc, 2017-12-25:
It is not in 2.2.1 It will be available in next version

PR #5563 (2017-12-25): 128x64 Display RAS when needed
Add display to non x7 ARM radio when XJTcd build-9x

[image]

This fixes #5520