opentx/opentx #5525: stickrev option missing from 'settings' for '9x with m128' in Companion 2.2

csurf, 2017-12-17:
I'm trying to upgrade my 9x w/ m128 chip and taranis gimbals to 2.2.*, but the 'stickrev' option is missing from the build options within Companion. Therefore, I can't download the correct build (if it's even available). Can someone please take a look at this?

[image]

kilrah, 2017-12-17:
I believe stickrev is only an option for m64, it's always built in for m128.

csurf, 2017-12-18:
I must be missing something then, because the openTX build that I downloaded & flashed through Companion doesn't allow me access the 'stick direction icons' within the main radio configuration menu. It simply skips over the 'channel order' option and jumps to the 'mode' option.

Is this simply a case of user error? How am I supposed to highlight and access the stick direction icons? There doesn't appear to be another 'hardware' screen to do the stick reversals either...

kilrah, 2017-12-18:
I have checked the code and it's indeed how it should be, enabled by default for m128. 
Unfortunately I don't think anyone in the team still has a 9x to test.

csurf, 2017-12-18:
Can you provide me with a direct link to a 9x-m128 build that should work so that I can test?

kilrah, 2017-12-18:
I don't even have a working build environment for AVR radios anymore... And since the problem is in the server's build process it wouldn't help fixing anything.

Any potential fix will have to be tested through the nightly builds.

csurf, 2017-12-26:
@kilrah 
how/where can i access the nightly builds for testing?

kilrah, 2017-12-26:
Nothing at all has been done with regard to this, so 2.2.1 release should be fine too...

closed 2019-08-19 by kilrah