opentx/opentx #5529: No joystick connection on windows 10 Taranis x9d+ 2.2.1

Toastwaffles, 2017-12-18:
So I went and updated my taranis x9d+ to the latest 2.2.1 firmware, and have no joystick connection. I was reading that people have to disable massstorage and cli, but i have neither of those checkboxes to uncheck. My buddy has been using his as a joystick for along time, but when he updated to the 2.2.1 from 2.1.9 his didnt work either. Anything that I could be missing. My brain is out of ideas.

Thanks. 

[image]

3djc, 2017-12-18:
Now a dynamic choice. Plug your cable in your radio and make a choice  (or store the choice in radio setup)

closed 2017-12-18 by 3djc

Toastwaffles, 2017-12-18:
Thank you so much