opentx/opentx #5532: openTX 2.2.1 - Horus12s - BT is not working

dk7xe, 2017-12-18:
Radio - Horus 12s
openTX - 2.2.1
Companion - 2.2.1

BT is not working on the Horus 12s. Have set BT to Telemetry and gave it a name.
According @3djc a hardware adresse should show up. But that's not present.
I have scanned BT with my SGS7 but the Horus is not seen.
[image]

3djc, 2017-12-18:
My bad. This has changed and on x10/12, that info is bellow master/BT in model setup. And BT is working according to our tests

closed 2017-12-18 by 3djc

kilrah, 2017-12-18:
Works, you may need to reboot the radio after changing the setting to telemetry.

dk7xe, 2017-12-18:
After reboot it showed up as available device but my android phone can not connect to the Horus. Saying no connection possible.
My windows computer is not recognizing the Horus at all  while it recognizes the Android mobile.

3djc, 2017-12-18:
Be aware that this is not a full BT module, only BLE. I likely requires a special app to deal with it

dk7xe, 2017-12-18:
Even BLE terminal https://play.google.com/store/search?q=BLE&c=apps&hl=de could not find it. But i give up now. ;)

kilrah, 2017-12-18:
The Free Link FrSky iOS app recognises it fine.

There is no app on android or Windows that can make use of it anyway at this point.

dk7xe, 2017-12-18:
ok, thank you!